rusefi/rusefi #4919: Cam showing on console but not TS trigger logger

ElDominio, 2022-12-29:
[image]

[image]

[image]

Have tried on E6 and same issue

ElDominio, 2022-12-29:
Problem solved, triggers inverted

My confusion arises from the idea of crank being primary no matter the trigger. I would suggest a fix but it would involve writing instructions and no one including me reads them

closed 2022-12-29 by ElDominio

rusefillc, 2022-12-29:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]