rusefi/rusefi #4927: TS Config comments trim at first new-line

nmschulte, 2023-01-02:
I noticed that the comments produced for TS are trimmed at the first new-line, due to this change:
https://github.com/rusefi/rusefi/commit/e50c67862367eb5972d8e104d2265f50b6948d1f#diff-46e1f984fe68d0a10452477abd40022fe895c9fe392a586dc2f2180752297a49

The parts of the comments seem helpful in TS, but this code change seems intended.  Can TS not support these comments, or does this logic get used for something other than TS as well?  Can we simply remove this change, or does something more need to occur?

rusefillc, 2023-01-02:
@nmschulte does linked change say
```
we might have detailed long comment for header javadoc but need a short field name for logs/rusEFI online
```
?

nmschulte, 2023-01-02:
@rusefillc I don't understand what that comment from the change means.  It seems to be saying "logs/rusEFI online" use the comment as the field name, but I'm unsure what that has to do with TunerStudio anyway.

rusefillc, 2023-01-02:
```
bit isDisabledByLua;AC disabled by Lua
```

[image]

nmschulte, 2023-01-02:
Mmm.  I see.  There's no way to split what feeds MLV/logs/rusEFI online and TunerStudio?  Some comments from `rusefi_config.txt` seem helpful but now are basically dead weight, whereas this trimming makes sense for `output_channels.txt` and `ac_control.txt` (and likely more).

rusefillc, 2023-01-02:
https://en.wikipedia.org/wiki/Moon_landing tells us that many things are possible with proper resource allocation.

``some comments`` reference is either your being playful or you assuming that I can read your mind or you trying to give me more work to figure out specific examples?

nmschulte, 2023-01-02:
This is just my observation from e.g. 5 ft pole -- perhaps I misunderstand the intent behind these comments.

Some specific examples, though pretty much any config in `rusefi_config.txt` with a new-line seems like it'd be useful to have the full comment in TunerStudio:

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L302
```
float baseFuel;Base mass of the per-cylinder fuel injected during cranking. This is then modified by the multipliers for CLT, IAT, TPS ect, to give the final cranking pulse width.\nA reasonable starting point is 60mg per liter per cylinder.\nex: 2 liter 4 cyl = 500cc/cyl, so 30mg cranking fuel.;"mg", 1, 0, 0, 500, 1
int16_t rpm;This sets the RPM limit below which the ECU will use cranking fuel and ignition logic, typically this is around 350-450rpm. \nset cranking_rpm X;"RPM", 1, 0, 0, 3000, 0
end_struct
```

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L358
```
	uint8_t dutyIfError;If an error (with a sensor, etc) is detected, this value is used instead of reading from the table.\nThis should be a safe value for whatever hardware is connected to prevent damage.;"%", 1, 0, 0, 100, 0
	uint16_t pwmFrequency;Select a frequency to run PWM at.\nSet this to 0hz to enable on-off mode.;"hz", 1, 0, 0, 500, 0
```

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L457
```
bit cj125isUaDivided;Is your UA CJ125 output wired to MCU via resistor divider? Ua can go over 3.3v but only at lambda >3, i.e very lean AFR above 44.1\nWhen exposed to free air and 17x gain, Ua will be 4.17 volt
```

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L462
```
bit cj125isUrDivided;Is your UR CJ125 output wired to MCU via resistor divider?\nLooks like 3v range should be enough, divider generally not needed.
```

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L471
```
bit displayLogicLevelsInEngineSniffer;Shall we display real life signal or just the part consumed by trigger decoder.\nApplies to both trigger and cam/vvt input.\n\nenable logic_level_trigger
```

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L481
```
int16_t tpsMin;Closed throttle, 1 volt = 200 units.\nSee also tps1_1AdcChannel\nset tps_min X;"ADC", 1, 0, 0, 1023, 0
int16_t tpsMax;Full throttle.\nSee also tps1_1AdcChannel\nset tps_max X;"ADC", 1, 0, 0, 1023, 0

int16_t tpsErrorDetectionTooLow;TPS error detection: what throttle % is unrealistically low?\nAlso used for accelerator pedal error detection if so equiped.;"%", 1, 0, -10, 0, 0
int16_t tpsErrorDetectionTooHigh;TPS error detection: what throttle % is unrealistically high?\nAlso used for accelerator pedal error detection if so equiped.;"%", 1, 0, 100, 110, 0
```

https://github.com/rusefi/rusefi/blob/master/firmware/integration/rusefi_config.txt#L524
```
engine_load_mode_e fuelAlgorithm;This setting controls which fuel quantity control algorithm is used.\nAlpha-N means drive by TPS commonly only used for NA engines\nSpeed Density requires MAP sensor and is the default choice for may installs\nMAF air charge is a cylinder filling based method that uses a mass air flow sensor.;
```

rusefillc, 2023-01-02:
At the moment I am very confused

https://github.com/rusefi/web_backend/issues/166

closed 2023-01-02 by rusefillc

nmschulte, 2023-01-02:
Awesome, thank you @rusefillc.

commit ba248475 (2023-01-02): TS Config comments trim at first new-line fix #4927
commit f1aa1d52 (2023-01-02): TS Config comments trim at first new-line fix #4927