rusefi/rusefi #4930: Toyota 1NZ-FE trigger (36-2CKP, 4-1CMP) issues

ElDominio, 2023-01-02:
The car runs well in semi sequential, but activating the cam input makes RPMs go insane and the car shut off.

https://rusefi.com/online/view.php?log=1134 -> Log shows this happening

[re_11.zip](https://github.com/rusefi/rusefi/files/10333643/re_11.zip) -> tooth log

ElDominio, 2023-01-08:
RPMs no longer go crazy, but cam position is jumpy

[OsciLogandNormalLogwithcapsRevE.zip](https://github.com/rusefi/rusefi/files/10368954/OsciLogandNormalLogwithcapsRevE.zip)
tunerstudio oscope and log

[re_15.zip](https://github.com/rusefi/rusefi/files/10368955/re_15.zip)
SD card tooth log

rusefillc, 2023-01-10:
Looks like https://rusefi.com/online/view.php?msq=1349&dialog=triggerConfiguration is the tune

mck1117, 2023-01-10:
invert the cam

rusefillc, 2023-01-10:
Ideal gap should be 3 on 36/2

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#362

We get a very clear gap of 2 that gives us a 15% spile in instant RPM which #4954 would be reporting "one day"

Something is not right @ElDominio 

[image]

mck1117, 2023-01-10:
It is more obvious to look at `triggerToothAngleError`, there is clear +10 then -10, which means one tooth is in the wrong spot.

[image]

Looking at the tooth log, you have the correct trigger type selected, but you need to __invert the primary trigger__.

mck1117, 2023-01-10:
[image]

ElDominio, 2023-01-10:
https://rusefi.com/online/view.php?msq=1364

Current tune on car

I will be with the car on Thursday

rusefillc, 2023-01-12:
crank seems happy now https://rusefi.com/online/view.php?log=1146

rusefillc, 2023-01-12:
@ElDominio says VVT is issue now so let's look at vvtSyncRatio

[image]

mck1117, 2023-01-12:
Something bad/weird is going on with the cam.

Top: crank (good)
Bottom: cam (wtf)

[image]

(by the way, it works great to open a trigger scope log from TS in megalogviewer!)

mck1117, 2023-01-12:
How is the cam wired? there's no connection between the cam and crank sensors, is there?

ElDominio, 2023-01-12:
Yes, there is, factory Toyota wiring shares the negative side of VR

I mentioned this on slack but aha slack is not GH lol

rusefillc, 2023-01-12:
@mck1117 I would love to learn new skills

@mck1117 at the moment I need simpler instructions to learn new skills sorry, as is I have no idea what new skills I can gain

mck1117, 2023-01-12:
> factory Toyota wiring shares the negative side of VR

Yeah, this isn't gonna work with the VR circuitry . Not sure what to do there. Any way you can run independent VR- wires for each?

mck1117, 2023-01-12:
> @mck1117 I would love to learn new skills
> 
> @mck1117 at the moment I need simpler instructions to learn new skills sorry, as is I have no idea what new skills I can gain

huh?

rusefillc, 2023-01-12:
"(by the way, it works great to open a trigger scope log from TS in megalogviewer!)"

this sounded like you were suggesting something but I have no idea what you are suggesting

ElDominio, 2023-01-12:
[image]

mck1117, 2023-01-12:
> "(by the way, it works great to open a trigger scope log from TS in megalogviewer!)"
> 
> this sounded like you were suggesting something but I have no idea what you are suggesting

nope, I've just learned this too! I opened it by accident with MLV and it worked great.

ElDominio, 2023-01-12:
Solution for VVT position has been established

Final problem: cranking RPM with inverted set to TRUE on primary trigger is not useable. I get it running by setting it to false, starting the engine, then switching to TRUE.

[2023-01-12_17.36.37.mlg.zip](https://github.com/rusefi/rusefi/files/10406421/2023-01-12_17.36.37.mlg.zip)

mck1117, 2023-01-12:
@ElDominio try grounding that wire while also connected to both cam/crank VR- inputs - just add a ground jumper between the VR- wire and sensor ground

ElDominio, 2023-01-13:
Connected jumper cable between sensor ground and VR-

Signal is significantly cleaner, but not very reliable.
Scopes of this at 
[bridgeoscope.zip](https://github.com/rusefi/rusefi/files/10414013/bridgeoscope.zip)

We will remove this bridge and run a seperate CMP+/- wire tor esolve issues, but I think VVT errors were due to sharing the VR

Also, noticed I duplicated this issue #4825

ElDominio, 2023-01-28:
I am commenting to document

It is recommended to add a 10nf cap across pins 4 and 5, and pins 2 and 4 to TLV3202AIDR for VR purposes

Pending testing

closed 2024-02-23 by ElDominio