rusefi/rusefi #4943: 12 tooth crank issues

ElDominio, 2023-01-06:
I have a 2JZ with this trigger setup using internal VR for both.

[suprare_12.zip](https://github.com/rusefi/rusefi/files/10357425/suprare_12.zip)

here is a cranking log with inverted trigger
[crankinglogenenero2023inverttrue.zip](https://github.com/rusefi/rusefi/files/10357431/crankinglogenenero2023inverttrue.zip)

adjacent tune
[crankingtoothlogsuprainvertedtrye.zip](https://github.com/rusefi/rusefi/files/10357436/crankingtoothlogsuprainvertedtrye.zip)

Changing to custom toothed wheel allowed it to sync, but sometime syncs on the wrong tooth (30 degrees out of timing)

rusefillc, 2023-01-09:
#4179

rusefillc, 2023-01-09:
https://rusefi.com/online/view.php?msq=1298 looks correct

rusefillc, 2023-01-09:
i see two issues on the log

1) vvt counter is not increasing. This setup absolutely depends on single tooth cam being alive

2) what's up with trigger gap? on the evenly spaces 12/0 crank I would expect 1, 1, 1, 1, 1ish equal gaps but that's totally not what the log shows? Some sort of electrical/VR issue? Which reminds me about build-on oscilloscope have you tried that?

[image]

ElDominio, 2023-01-09:
[image]

we can see cam+1 is extremely close to one of the crank teeth

Trigger scope pic with 
custom tooth wheel 12 0 missing
both sensors VR, boards has 10nF caps soldered on

[logsoncustom12toothnomissing.zip](https://github.com/rusefi/rusefi/files/10376518/logsoncustom12toothnomissing.zip)

customtrigger12minus0yessttart.csv <- started with custom tooth wheel 12 0 missing, oscope log, other log is the same but i had injection disabled

[12toothcrank scranking scope.zip](https://github.com/rusefi/rusefi/files/10376526/12toothcrank.scranking.scope.zip)

only a scope of cranking since this trigger does not start

ElDominio, 2023-01-09:
Using these tooth gaps
[image]

Fired right up, will check for possible sync problems
[running12toothwithcustomgaps.zip](https://github.com/rusefi/rusefi/files/10376628/running12toothwithcustomgaps.zip)

rusefillc, 2023-01-09:
@ElDominio says "Toyota wheel is unusual it gives gap numbers during cranking"

[image]

rusefillc, 2023-01-09:
2JZ would be global trigger offset 65 but same wheel could be Honda, not hard coding for now

[image]

commit 1b080bcd (2023-01-09): 12 tooth crank issues fix #4943
commit 00aef081 (2023-01-10): 12 tooth crank issues fix #4943
closed 2023-01-09 by rusefillc