rusefi/rusefi #4949: sensorinfo does not contain physical pin info

rusefillc, 2023-01-07:
analoginfo used to mention HW channel
```
2021-04-01_17_14_38_244: EngineState: CLT ADC13 ADC_SLOW PC2 adc=0.82/input=1.65v/divider=2.00
2021-04-01_17_14_38_244: EngineState: IAT ADC14 ADC_SLOW PC3 adc=0.98/input=1.97v/divider=2.00
```

see #2517

It looks like #4646 has wiped that info out

mck1117, 2023-01-07:
This?

[image]

rusefillc, 2023-01-07:
No, that

[image]

rusefillc, 2023-01-07:
also this @mck1117 

[image]

mck1117, 2023-01-07:
eh? the command `sensorinfo` first shows all ADC channels in use, then shows all sensors configured

rusefillc, 2023-01-07:
Ohh sorry

closed 2023-01-07 by rusefillc