rusefi/rusefi #4953: Ignition IAT corr IAT error what

ElDominio, 2023-01-07:
[image]

ElDominio, 2023-01-07:
[image]

rusefillc, 2023-01-07:
@ElDominio did you full erase on firmware update and restored settings from TS?

looks like you did not full erase so new table was using random left-over bytes?

ElDominio, 2023-01-07:
Problem is/was? IAT values in the lower two rows were negative. Although negative air temp is not possible in Puerto Rico, it might be somewhere else?
[image]

rusefillc, 2023-01-07:
@mck1117 it's a typo
```
		ensureArrayIsAscending("Ignition IAT corr IAT", config->ignitionIatCorrLoadBins);
		ensureArrayIsAscending("Ignition IAT corr Load", config->ignitionIatCorrLoadBins);
```

ignitionIatCorrTempBins not validated while ignitionIatCorrLoadBins is validated twice.

closed 2023-01-10 by rusefillc

rusefillc, 2023-01-17:
@ElDominio reminder to use more words. I see nothing broken, I see you not wiping chip blank and @mck1117 not adding CHANGELOG about the incompatible configuration change

please manually fix vertical axis