rusefi/rusefi #4955: totalFuel and totalIgnition correction gauges in Tunerstudio

ElDominio, 2023-01-08:
Having a "totalX" gauge for corrections ccan help catch corrections that might be driving a tuner crazy.

Also, a view like this would help but I don't know about the difficulty of making this
[image]

Shows each correction individually and then the total, which helps catching tuning mistakes. Every correction should ideally have a gauge

rusefillc, 2023-01-18:
@ElDominio more of fuel and timing data points were extracted into corresponding "view" section - see fuel and ignition state in "view"

Also any "view" entry could now be a gauge with little effort of my end. See new subcategories for gauges, let me know if we are on the right track.

Any additional "view" metric to expose as a gauge please let me know.

closed 2023-01-18 by rusefillc

ElDominio, 2023-01-18:
Thank you very much, will test soon