rusefi/rusefi #4975: Cam-as-primary is vague

ElDominio, 2023-01-17:
Why are some triggers designed to use cam as primary? config should always be crank as primary to eliminate vaugeness

Right now 4G63 uses Cam as primary, but 95 Dodge Neon doesn't or does? If it doesn't, why not? Cam is part of trigger shape  @rusefillc

ElDominio, 2023-01-17:
[image]

In one example, cam is primary

In other example, cam is not even considered.

@rusefillc says "please follow images" which implies to follow order on images from left to right

Sentence is not in "all supported triggers" page

rusefillc, 2023-01-17:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#norman-note

ElDominio, 2023-01-17:
Note helps significantly

New users don't know who Norman is though

closed 2024-02-23 by ElDominio