rusefi/rusefi #4976: Dodge 1995 trigger reading double RPM

ElDominio, 2023-01-17:
log and tune at https://rusefi.com/online/view.php?log=1147

[image]

rusefillc, 2023-01-17:
@ElDominio please help me understand what's going on

Do you see the blue warning note?

[image]

How does blue warning correspond to your selection?

[image]

rusefillc, 2023-01-17:
@ElDominio i am _really_ worried overall

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#dodge-neon-1995

ElDominio, 2023-01-17:
https://rusefi.com/online/view.php?log=1148

settings as in trigger page now with log

mck1117, 2023-01-17:
should this not get converted to a symmetrical crank trigger? is the "crank" pattern actually on the crank, or are they all on the cam "CAS-style"?

ElDominio, 2023-01-17:
crank is actually on crank

mck1117, 2023-01-17:
> crank is actually on crank

OK, in that case, it must get converted. Right now you'll get timing inaccuracy since some teeth are on cam, and some are on crank, but we treat them as phase locked (which they aren't, if the belt stretches then some of your events move around)

ElDominio, 2023-01-21:
This is a log from the guy who is running the car

Apparently, started it with timing retarded (he changes trigger angle from 140 (which is actual zero) to 170 and the car starts, he then changes back to 142 and it runs well.

[LogCranking170triggerFixed0cranking#2.zip](https://github.com/rusefi/rusefi/files/10471587/LogCranking170triggerFixed0cranking.2.zip)

This is a log at 135 trigger angle, doesn't start. Might be something that stating while severly undertimed makes it start correctly?
[LogCranking135trigger-30cranking.zip](https://github.com/rusefi/rusefi/files/10471589/LogCranking135trigger-30cranking.zip)

mck1117, 2023-01-21:
OK, this requires a tooth log of some form to proceed. Logic analyzer, SD card, whatever, but something is *VERY* wrong with the trigger pattern and it's not worth our effort to try and diagnose this over 3 hops of the game of telephone. Get a tooth log of the same thing as the "135 trigger 30 cranking" log and then we can make progress.

ElDominio, 2023-01-23:
[image]

Did a tooth log to check how and when it loses sync to try to understand why

mck1117, 2023-01-23:
> when it loses sync to try to understand why'

this is of no use, we need to change how this trigger mode works in code first

mck1117, 2023-01-24:
apparently the engine has blown up, but:

our trigger pattern was totally wrong. It's close, but the teeth weren't actually spaced right.

[image]

with #5003 the pattern is now within a degree or so of correct.

rusefillc, 2023-01-25:
Have we blown up the engine?

mck1117, 2023-01-25:
> Have we blown up the engine?

@ElDominio initial assessment was no

ElDominio, 2023-01-25:
Today they tore it open, oil pump failure lol

mck1117, 2023-01-25:
> oil pump failure

ouch

closed 2023-01-25 by rusefillc