rusefi/rusefi #4987: [EPIC PROPOSAL] Board-specific repos in charge of building their own firmware, generating .ini, etc.

chuckwagoncomputing, 2023-01-21:
## What if... Each board had its own Github repository

Github Actions on that repo could be responsible for building firmware for that board, generating config files for that board, etc.

## Pros

- This repo wouldn't be full of random board stuff that's never used, but takes up CI time
- Quicker developer turn-around as all actions will finish quicker

## Cons

- Is it possible?
- I'm sure there's a ton of stuff I haven't thought of. This has been in the back of my mind for a long time, but this project is too big for me to understand all the implications.

## How?

We could have central hosting for the workflow files that those would need, with each job in its own repo, and the board repos 'using' them in their workflows.

eg. We would have a repo rusefi/board-firmware-action
and mck1117/proteus would have a 'uses: rusefi/board-firmware-action' step

so that if we needed to update the firmware build process, it would be done in one location.

I haven't thought this through entirely, but I have some hope that we could get by without having rusefi/rusefi as a submodule that would need to be updated - The actions would just fetch the repo as needed.

## Open Questions

- How do we signal that a firmware change has taken place and boards need to build the firmware?
  Possibly with https://github.com/marketplace/actions/repository-dispatch

- How do we notify the person who opened the PR or pushed to master that their change broke something for a specific board even though it didn't for the main firmware build?
  The rusefi/board-firmware-action could post it's status to some simple server, which could  be checked by a workflow on rusefi/rusefi which would just wait until all boards returned a status
  However - that would take up one "concurrency slot"

## Feedback

Please tell me your thoughts and concerns, and I can add them to this.

mck1117, 2023-01-21:
> This repo wouldn't be full of random board stuff that's never used, but takes up CI time

If it's really never used, we should delete that stuff.

> Quicker developer turn-around as all actions will finish quicker

Hosted builds already run in parallel, so skipping some and/or just making them faster will help. We can also pay some amount of money to increase the parallelism if we care enough, or host our own machines to do it instead.

> I'm sure there's a ton of stuff I haven't thought of.

- How do snapshot builds get uploaded? We do really want a snapshot build available for every commit for every board that people actually use.
- In case of some "broad" change that affects all boards (for example, what if we renamed `injectionPins` to `fuelThingyPins`)? How do you make an atomic change to all board build/defaults configs at once? (the current model also enforces that boards _cannot_ get out of date when something like that changes. If the main repo is gated on all boards working, you end up deadlocked if some board's owner doesn't want to or can't edit their repo).
- How do developers know that their change to the main repo broke something on board X?
- Is it a good idea for "official" builds to then use some separate build process?

chuckwagoncomputing, 2023-01-21:
In reply to @mck1117 
> If it's really never used, we should delete that stuff.

maybe not *never* used - I'm thinking stuff like the cypress firmware build

> skipping some

The more we skip, the less we know if we broke something

> making them faster

Yes please

As far as CI is concerned, this is just a complicated way of increasing the parallelism.
But more boards/projects all going through one repo also increases @rusefillc's workload.
e.g. merging a pull request for a connector pinout for a random board (or even [not-a-board](https://github.com/rusefi/rusefi/pull/4969))
To some extent, that's a result of his approach of being the sole person in control of this project, but that's a separate discussion that I personally don't want to get involved with.

> How do snapshot builds get uploaded? We do really want a snapshot build available for every commit for every board that people actually use.

I see no reason that the hypothetical rusefi/build-firmware-action couldn't do that. If you wanted maximum security, you could give each board owner a user account on the rusefi server, with write access to only one directory. I don't think that's possible with the current host though.

> In case of some "broad" change that affects all boards (for example, what if we renamed injectionPins to fuelThingyPins)? How do you make an atomic change to all board build/defaults configs at [...]
How do developers know that their change to the main repo broke something on board X?

I addressed these in the "Open Questions" section, and give some ideas for solving them.
Obviously, it would be possible for a board owner to not set up their repo correctly with the workflow file, secrets, etc. and then their firmware builds wouldn't work. I think adding repos the list that the main build is gated on would be a manual process.

> Is it a good idea for "official" builds to then use some separate build process?

Do you mean the "default" firmware build, or "official" boards like MRE?

mck1117, 2023-01-21:
> > making them faster
>
> Yes please

looks like right now a "gate" firmware build is:
- ~2:00 git checkout
- ~1:30 install dependencies
- 1:10-1:30 build firmware

Hosting our own machines would knock off 74% of the time, roughly, plus would increase the parallelism to as many as we want.

nmschulte, 2023-01-21:
Pre-built images (i.e. deps, setup) would also help knock off some time.  If the git checkout isn't shallow that could save a lot too.

rusefillc, 2023-01-21:
i have to preserve my resources, I cannot afford to have this issue visible on the list.

Anyone with capacitary to _help_ please consider PRs in the direction of LTS https://github.com/rusefi/rusefi/issues?q=is%3Aissue+is%3Aopen+lts+label%3ALTS

closed 2023-01-21 by rusefillc

mi-hol, 2023-01-26:
@chuckwagoncomputing I'd be in favor of re-opening this issue and adding this action to a new roadmap.
(see my very similar issue https://github.com/rusefi/rusefi/issues/4970)

The current process is wasting so many resources and therefore takes such a long time to complete!

mi-hol, 2023-01-26:
acceptable: 1 min
unacceptable: >2 min

mck1117, 2023-01-26:
> acceptable: 1 min unacceptable: >2 min

I think that's just not possible. The actual compilation step is in the 1m30s range right now.

mi-hol, 2023-01-26:
why should a goal be limited by what is the current setup?
Sometimes goals are required to finally "go to the moon"

mck1117, 2023-01-26:
but what value does 1 minute provide over 5 minutes?

I'm not sure "going to the moon" means 1 minute PR build gates, but OK.

mi-hol, 2023-01-26:
https://github.com/rusefi/rusefi/pull/5005 does count for reaching the goal, right?

mi-hol, 2023-01-26:
https://github.com/rusefi/rusefi/issues/3618 is related to this too and apparently was closed successfully via https://github.com/rusefi/rusefi/pull/3622

chuckwagoncomputing, 2023-01-26:
This issue is closed, please let it rest for now. In the meantime we can work on doing things that make it less necessary.
#5005 and #3622 help make this less necessary.
#3622 was closed, not merged.
I am working on moving interactive pinout generation to its own action that can be called by any repo.
https://github.com/chuckwagoncomputing/interactive-pinout
You can help with this issue by moving your mappings that don't need a TunerStudio config to their own repo and using that action.