rusefi/rusefi #4992: LTS: folder inside bundle and self identifition

rusefillc, 2023-01-22:
At the moment all .zip bundles have files within "snapshot_timestamp" folder. While we need a way to say what bundle we look at, current approach is broken

Timestamp in folder makes no sense in ligh of bundle autoupdate, looks like we shall remove timestamp

For lts branches shall we use branch name as folder name?

Open question if we need some metainfo.txt file with original bundle timestamp, and maybe separate entry somewhere with latest auto update timestamp.

rusefillc, 2023-01-22:
@chuckwagoncomputing any interest?

mck1117, 2023-01-22:
should the folder name just be the signature?

right now it's like `snapshot_20230111_174746_proteus_f4_rusefi`

it could be like `rusEFI master.2023.01.22.proteus_f4.2951145916`

rusefillc, 2023-01-22:
With autoupdate workglow original bundle timestamp should we much less visible imho

chuckwagoncomputing, 2023-01-25:
So something like `rusEFI master.proteus_f4`?

rusefillc, 2023-01-25:
Yes

Minor detail let's call main/master "snapshot" looks like that's the client facing term we use

PR #5008 (2023-01-28): [LTS] Put branch in bundle name (please read caveats)
This will fix #4992 and #4993.
- Removed date from bundle name
- I chose the approach of using the name of the folder rather than putting the info in bundle_name.txt
  I chose this because I thought it would be simpler, but the exception to that is switch_bundle.sh, which got a little more complicated - see Caveats below. I'm not set on doing things this way.
- The bundle names are `rusefi.<branch name>.<bundle>`
  If the branch is master, then the `<branch name>` is "snapshot".
  Examples:
  `rusefi.snapshot.proteus_f7`
  `rusefi.lts_foo.m [... cut]
closed 2023-01-28 by rusefillc