rusefi/rusefi #4997: rusefi_bundle_f429-discovery bundle not working on my f429-discovery

rusefillc, 2023-01-22:
I've just flashed snapshot_20230122_032226_f429-discovery_rusefi into an f429-discovery board and no life :(

nmschulte, 2023-02-13:
I have only a single Proteus board.  At the moment I'm not interested to own any other boards but I'm happy to receive any for dev/testing.

FYI `f429-discovery` has other build issues after resolving `STM32_LSECLK`:
```
ChibiOS/os/hal/ports/STM32/LLD/USARTv1/hal_serial_lld.h:339:21: error: type of 'SD6' does not match original declaration [-Werror=lto-type-mismatch]
  339 | extern SerialDriver SD6;
      |                     ^
ChibiOS/os/hal/ports/STM32/LLD/USARTv2/hal_serial_lld.c:118:14: note: 'SD6' was previously declared here
  118 | SerialDriver SD6;
      |              ^
```

rusefillc, 2023-02-13:
Cute. Any chance you could have a look?

nmschulte, 2023-02-13:
I am looking -- it seems like USARTv2 should not be included; I'm not sure if this is a ChibiOS bug or something about rusEFI:
https://github.com/rusefi/ChibiOS/blob/6b154adfaa56afe7570a19bbce552d2e7ea9b891/os/hal/ports/STM32/STM32F4xx/platform.mk#L44

Can you direct?  I'm not sure where to look about rusEFI for any discrepancy.

mi-hol, 2023-08-15:
@rusefillc is this still an open issue for f429-discovery board ?