rusefi/rusefi #4999: Refactoring: less confusion macro names around serial/uart/etc in connectivity

rusefillc, 2023-01-23:
Current terminology could be improved

rusefillc, 2023-01-23:
First order of business
We have "serial" and "uart" what is the proper term for "serial or uart"?

mck1117, 2023-01-23:
For context, which defines?

rusefillc, 2023-01-23:
```
HAS_PRIMARY
HAS_SECONDARY
```

nmschulte, 2023-01-23:
The console filenames say 'serial', but UART seems less ambiguous in the bigger scope: but this is limited to the console§TS IO where serial isn't really ambiguous.

dron0gus, 2023-01-23:
ChibiOS has both Serial and UART drivers for same IP block in STM32. Drivers provide different functionality. So serial != uart. At least in ChibiOS therms.

rusefillc, 2023-01-23:
Serial is uSart, I am thinking to use UxART as umbrella term unless there are better ideas

mck1117, 2023-01-23:
> Serial is uSart, I am thinking to use UxART as umbrella term unless there are better ideas

what is the problem we are trying to solve?

rusefillc, 2023-01-23:
Has_primary is a shitty variable name

mck1117, 2023-01-23:
BTW those particular defines are _not_ defined in any config file, they're defined as a function of others in tunerstudio_io.h

[image]

mck1117, 2023-01-23:
#3578 already improved this by not having separate serial vs. uart defines, and just deciding based on the type of whatever thing listed there

rusefillc, 2023-01-23:
Here we will only rename, nothing else.

commit c1fea618 (2023-01-23): Refactoring: less confusion macro names around serial/uart/etc in connectivity fix #4999
closed 2023-01-23 by rusefillc