rusefi/rusefi #5000: 4G93 only runs with "4G93 Only Cam", not "4G93 both"

ElDominio, 2023-01-23:
the car runs on C7 Cam hall input just fine using "4G93 only cam". The problem is using the "4G93 both fronts" which  4 tooth crank (8 because both edges), which is wired to C3.

logic analyzer log 
[4g93start.zip](https://github.com/rusefi/rusefi/files/10478001/4g93start.zip)

rusefillc, 2023-01-23:
Need tune link, need TS log link

ElDominio, 2023-01-23:
https://github.com/rusefi/rusefi/issues/4918#issuecomment-1374636896

mck1117, 2023-01-23:
are both signals physically mounted inside the same assembly? CAS with two rings of holes like an NA miata?

rusefillc, 2023-01-23:
> are both signals physically mounted inside the same assembly? CAS with two rings of holes like an NA miata?

No

mck1117, 2023-01-23:
> > are both signals physically mounted inside the same assembly? CAS with two rings of holes like an NA miata?
> 
> No

Right now the trigger expects the cam+crank to have "locked" phase like an NA miata - that won't work here.

I'll PR momentarily.

mck1117, 2023-01-23:
@ElDominio what is the exact year/model of where this engine came from

rusefillc, 2023-01-23:
```
s->useOnlyPrimaryForSync = addSecondary;
```
@mck1117 please elaborate what you mean. It looks like both runs with useOnlyPrimaryForSync  = true?

mck1117, 2023-01-23:
If the two channels are physically not connected (ie, one is physically mounted to the crank and the other to the cam) we can't ensure that they're timed correctly, aka we have to decode using "vvt" instead of a single trigger

all other options are imprecise re: ignition timing etc

rusefillc, 2023-01-23:
@mck1117 what absolute value of deviation between channels do we expect? 10 degree? 5 degree of crank?

While obviously those happen I wonder if such a (minor) deviation would directly cause "does not run on 4G93 both".

mck1117, 2023-01-23:
> > are both signals physically mounted inside the same assembly? CAS with two rings of holes like an NA miata?
> 
> No

wait, is that true?

[image]

ElDominio, 2023-01-23:
1999 Mitsubishi Mirage, does NOT use that CAS @mck1117

ElDominio, 2023-01-24:
Only thing left is converting to use both ends of trigger, like so 
[image]

rusefillc, 2023-01-27:
@ElDominio please clarify current state, does the car run satisfactory now?

rusefillc, 2023-01-27:
PS: please provide logs

ElDominio, 2023-01-27:
I do not own the car and  the installer handed it in, I'm trying to get it back to test but no news currently

ElDominio, 2023-02-16:
Got it at the shop, available for tests.

Current method of custom toothed wheel of 2 teeth + 4G63 VVT setting is not working. Also, there is a VVT 4g63 that uses a different trigger pattern (with real VVT), so i suggest a name change for this trigger (4G63 No VVT)

Log of car cranking:
[Uploading split cam crank.zip…]()

4G93 cam only is broken now

rusefillc, 2023-02-16:
@ElDominio need help https://github.com/rusefi/rusefi/issues/5092

ElDominio, 2023-02-17:
[split cam crank.zip](https://github.com/rusefi/rusefi/files/10761948/split.cam.crank.zip)

rusefillc, 2023-04-01:
Too many tickets closing this one @ElDominio

closed 2023-04-01 by rusefillc