rusefi/rusefi #5015: LTS: firmware matcher needs to learn to migrate

rusefillc, 2023-01-30:
rusEFI console autoupdate feature has protection against accidental update of for instance MREF4 binary into proteusF7

At the moment that logic does not handle old vs new bundle name, also how would it handle branch switch?

[image]

rusefillc, 2023-01-30:
@chuckwagoncomputing you said you java now? :)

rusefillc, 2023-01-30:
``proteus_f7`` comes as bundle part of ``#define TS_SIGNATURE "rusEFI master.2023.01.30.proteus_f7.3726161684"``

We need a new terminology I guess ``proteus_f7`` is bundle TYPE and ``rusefi.snapshot.proteus_f7`` is FULL bundle name?

chuckwagoncomputing, 2023-01-30:
Re: Terminology
I suggest `proteus_f7` is "bundle target"

rusefillc, 2023-01-30:
Separate issue #5016

I am making progress on this one here hope to push soon.

commit 6522bd29 (2023-01-30): LTS: firmware matcher needs to learn to migrate
fix #5015
closed 2023-01-30 by rusefillc