rusefi/rusefi #5016: LTS: all vs stm32f407 mess

rusefillc, 2023-01-30:
We have technical debt/historical mess around "default" bundle

* bundle subfolder is rusefi.snapshot.stm32f407
* signature says "all" ``#define TS_SIGNATURE "rusEFI master.2023.01.30.all.2319868418"``
* java has custom code

```
        RusEfiSignature s = SignatureHelper.parse(signature);
        String bundleName = BundleUtil.readBundleFullName();
        if (bundleName != null && s != null) {
            String signatureWithPrefix;
            if ("all".equals(s.getBundle())) {
                signatureWithPrefix = PREFIX;
            } else {
                signatureWithPrefix = PREFIX + "_" + s.getBundle();
            }
```

Does discovery auto-update work at the moment? Shall we reduce the amount of technical debt?

mck1117, 2023-01-31:
maybe it's best for `all` to go away, since it isn't actually `all`

chuckwagoncomputing, 2023-01-31:
yep, I'm working on this. I'm aggressively removing stuff and finding out what happens.
https://github.com/rusefi/rusefi/compare/master...chuckwagoncomputing:rusefi:yeet?expand=1

PR #5018 (2023-02-03): Replace "primary" and "default" builds with "f407-discovery"
It has been asked whether the primary build serves any purpose, and there hasn't been any answer. Therefore we shall yeet with extreme prejudice.

- Remove "rusefi_bundle"
- Remove "rusefi_bundle_default"
- Introduce "rusefi_bundle_f407-discovery"
- Reset one more file after board-specific gen-config
- Get rid of a gen_config_default call of unknown purpose (maybe for #3367)
- Don't check for "all"
- Don't run gen_config_default in gen_config

This should fix:
#5016
#3182
#3085
#3367