rusefi/rusefi #5036: Antilag bug

Benjamin-neo, 2023-02-06:
Antilag
Antilag activation by pin state always on no. (test on different pin, pullup, pulldown)
LaunchControl have same fonctionnality and work.

rusefillc, 2023-02-06:
@Benjamin-neo is this tested on bench or real car? Any content you can share build thread pictures logs videos?

Benjamin-neo, 2023-02-06:
tested on car, and on bench to tested on different pin to known if problem is hadrware or firmware. have look to compare with lauch same fonction, bug is in only on antilag ALSactivate pin state doesn't change .I dont tell antilag doesnt work , I report bug only in antilag activation by switch pin, all work expect ALSactivationPinstate

rusefillc, 2023-02-06:
@Benjamin-neo glad to hear tested on car, looking to see more info on the forum

rusefillc, 2023-02-06:
@Benjamin-neo will look into your bug report

Benjamin-neo, 2023-02-06:
@rusefillc yes will post more on forum when project will ready.
For the bug, I have look at more, the problem is in activation mode ( switch Input/ always active). it seem to be simple inverted. maybe just need correct .ini
I let you see, I hope this info will help you

rusefillc, 2023-02-07:
@TurboMarian looks like ALSActivatePinMode field was never used?

rusefillc, 2023-02-07:
@Benjamin-neo I've made a change please try https://rusefi.com/build_server/ no sooner than 20 minutes from now :)

Benjamin-neo, 2023-02-07:
@rusefillc 
I have look at and test Turbomarian , here is solution for switch antilag
[image]

Benjamin-neo, 2023-02-07:
Have test f407 from bundle, doesnt work.
test with previous version and just != instead of == . and work perfectly

rusefillc, 2023-02-07:
Well I do not get it

ALSActivatePinMode was not used I assume it's wrong for ALSActivatePinMode to be not used?

Also the != or == please someone explain in human words

Benjamin-neo, 2023-02-07:
yes it's wrong for ALSActivatePinMode.
In file [firmware/controllers/algo/antilag_system.cpp]
turbomarian has only made on change == to !=

isALSSwitchActivated = engineConfiguration->antiLagActivationMode != SWITCH_INPUT_ANTILAG;

Just this edit solve the problem, als can be use with switch activation, pin is correctly selected, pullup pulldown , invert.. all this thing work . test on bench on multiple pin.
Als fonction is better on switch instead of always. als it's not safer for turbo and exhaust

rusefillc, 2023-02-08:
@TurboMarian i have a feeling that "!=" random change was not addressing root cause of the issue. Clearly the code was intended to have "=="

```
typedef enum __attribute__ ((__packed__)) {
	SWITCH_INPUT_ANTILAG = 0,
	ALWAYS_ON_ANTILAG = 1,
} antiLagActivationMode_e;
```

```
    #define antiLagActivationMode_e_enum "Always Active", "Switch Input"
```

root cause: .ini file texts do not match enum values (always value "1" in enum versus value "0" in ini)

@TurboMarian please please please get to root causes not lucky hacks

rusefillc, 2023-02-08:
@Benjamin-neo please update firmware and update TS project to load fixed .ini file

closed 2023-02-08 by rusefillc

Benjamin-neo, 2023-02-08:
Hi, just test it. switch mode like it do. just a last little bug , you can see on screen, while switch is select we can set pin , pull mode, invert, we can setting while select active always. test on last bundle f407 and .ini dowload with bundle. you can see sw on screenshot
[image]

rusefillc, 2023-02-09:
@Benjamin-neo plz try latest .ini file (required latest firmware just to get latest signature)

commit a262df4d (2023-02-07): Antilag switch pin mode was ignored fix #5036
commit 620af07b (2023-02-08): Antilag switch pin mode was ignored fix #5036
commit c3a3b4e7 (2023-02-09): Antilag switch pin mode was ignored fix #5036