rusefi/rusefi #5045: Is cranking fuel enrichment working correctly?

ElDominio, 2023-02-07:
[image]

Doesn't show as part of the total fuel multiplication?

rusefillc, 2023-02-07:
cranking and running fuel is calculated totally differently. please raise your cranking limit, please look at cranking gauges which we probably do not have let me look :)

ElDominio, 2023-02-07:
Cranking limit is 400RPM

rusefillc, 2023-02-07:
@ElDominio I now suspect that we have crap in crap out situation, specifically I have no idea which correction you are talking about since stuff is truncated. based on bad info and bad brain I produce bad answers

ElDominio, 2023-02-07:
[image]
This is the table is shown in the screenshot

[image]
Current cranking settings

rusefillc, 2023-02-07:
@ElDominio please help by mentioning CLT in ticket title

ElDominio, 2023-02-07:
Done

ElDominio, 2024-02-23:
says done right there so i'm going to believe past me

closed 2024-02-23 by ElDominio