rusefi/rusefi #5060: auto-calibrated ETB overheats TLE9201

rusefillc, 2023-02-13:
same issue on 4chan rev C and latest 4chan rev F
EXP-1270 small battery maybe it's discharged note VBatt going doing

https://rusefi.com/online/view.php?msq=1393

[ETB-overheat.zip](https://github.com/rusefi/rusefi/files/10718099/ETB-overheat.zip)

[image]

rusefillc, 2023-02-13:
added regulated power supply to the battery
swapped this VAG ETB for same VAG ETB

same result - 4chan rev F is overheating

same

rusefillc, 2023-02-13:
happy Proteus F7 setup

https://rusefi.com/online/view.php?msq=1394

log https://rusefi.com/online/view.php?log=1165

[image]

rusefillc, 2023-02-13:
Hold on, the happy log and perfect ETB above is with DEFAULT PID settings https://rusefi.com/online/view.php?msq=1394&dialog=etbDialog

I've hit ETB auto-tune and only gave it a second to run. BOOM! I got lame PID settings which have me insane ETB duty rate of change and I got Proteus h-bridge shut down on me.

Next I run ETB auto-tune again this time longer say 10 seconds. And now Proteus has legit healthy PID settings.

rusefillc, 2023-02-13:
Ok let's retrace steps

4chan, manually reverting PID settings to default.

https://rusefi.com/online/view.php?msq=1395

same (?) PID settings behave way worse on 4chan comparing with Proteus - that is hopefully nicely documented now

[2023-02-12_23.45.44.mlg.zip](https://github.com/rusefi/rusefi/files/10718868/2023-02-12_23.45.44.mlg.zip)

Allow me not to document but it looks like 4chan is unable to get nice ETB pid regardless how long I run auto tune.

rusefillc, 2023-02-13:
Ohh @ElDominio confirms stuff https://github.com/rusefi/alphax-4chan/issues/100

mck1117, 2023-02-13:
Why is vbatt changing so much? this suggests that you have a power supply problem.

closed 2023-09-10 by rusefillc