rusefi/rusefi #5061: ETB auto-calibrate usability: both buttons should not be enabled at the same time

rusefillc, 2023-02-13:
At the moment both "start ETB pin autotune" and "stop ETB pin autotune" buttons are always or often enabled at the same time.

Would be nice for buttons to know current state of the process. If we are NOT tuning - start should be enabled, stop should be disabled

while we are tuning start should be DISABLED and stop should be ENABLED

[image]

rusefillc, 2023-02-13:
@nmschulte is this by any chance something you could help with?

nmschulte, 2023-02-13:
On it; free to assign to me.

rusefillc, 2023-02-13:
@nmschulte you would need to expose "isAutoTuneRunning" bit output for UI to use it for conditional logic

see into existing .txt files maybe we even have the bit already. maybe only UI change is needed

nmschulte, 2023-02-13:
Cool.  While researching, I found this, semi-related ...
https://github.com/rusefi/rusefi/blob/master/firmware/controllers/actuators/electronic_throttle.cpp#L1036

which lead to this (ex. `isFanOn`) ... is this file still deprecated; what's the alternative?
https://github.com/rusefi/rusefi/blob/master/firmware/console/binary/output_channels.txt#L4

rusefillc, 2023-02-13:
https://github.com/rusefi/rusefi/commit/ce3cfc89c19286b16b1e9790d6ac30f457b8a678

rusefillc, 2023-02-13:
DC vs ETB is a much much bigger fish to fry :(

nmschulte, 2023-02-13:
It looks like `calibrationMode` already outputs the status, so that's a win.

"DC" means "deprecated" or ?

rusefillc, 2023-02-13:
DC as https://en.wikipedia.org/wiki/DC_motor

like DC motor wastegate code flow should probably NOT go through electronic_throttle.cpp right?

closed 2023-02-13 by rusefillc