rusefi/rusefi #5073: inclusion of PCAN dll violates GPLv3 license

mck1117, 2023-02-13:
Per the GPL, everything distributed as "rusefi" must have source available for users to recreate all binaries by themselves.

No source is available for `PCANBasic.dll` and `PCANBasic_JNI.dll`.

This means it is not allowed to:

- Distribute GPL code that links against this library
- Distribute this library as part of a GPL package

rusefillc, 2023-02-13:
@mck1117 please clarify why do you focus on PCAN specifically? Do we have source code for libgcc_s_seh-1.dll?

mck1117, 2023-02-13:
> libgcc_s_seh-1.dll

https://github.com/gcc-mirror/gcc/blob/master/libgcc/libgcc2.c

> STM32_Programmer_CLI.exe

Calling something via `exec` (ie, launching external program, passing cmdline args like we do to invoke `STM32_Programmer_CLI.exe`) doesn't cause GPL to be invoked. So long as their license allows redistribution and we only link it via running that separate program, this part is OK.

Loading a DLL directly from your process (`PCAN*.dll`) DOES trigger the provisions of the GPL.