rusefi/rusefi #5075: Nick wonders if DC wastegate is still FUBAR

rusefillc, 2023-02-13:
%subj%

rusefillc, 2023-02-14:
@Abricosvw 

[image]

```
mapSensor = Sensor.new("map")
mapSensor : setTimeout(3000)

function onTick()
	local tps = getSensor("TPS1")
	tps = (tps == nil and 0 or tps)
	mapSensor : set(tps)
end
```

rusefillc, 2023-02-14:
[image]

rusefillc, 2023-02-14:
[image]

rusefillc, 2023-02-14:
Status update:
just fixed percent scaling issue https://github.com/rusefi/rusefi/commit/254c16f5bc368cc2e4f8b4100175eccdce3ded5a

As is now code works with just one pid directly from MAP into DC duty cycle, that's the result of Jan 2023 attempts to get any life out of this functionality which seems to have been very broken

One PID meaning we totally ignore wastegate position which is probably not a good sign we would be overheating h-bridge if target could not be met

Is it that two PIDs control with position is needed just to avoid overheating or else? As is I am pretty confused and can use some help here.

@mck1117 @ElDominio

rusefillc, 2023-02-18:
steps back towards #4965

rusefillc, 2025-05-29:
@Abricosvw what's the status?

Abricosvw, 2025-05-30:
it looks like work ...
but 
need ASAP auto PID for DC motor ...
without it people won't be able to set up pid on their own and H-bridge chips will just burn out