rusefi/rusefi #5091: rusefi online does not have .ini for f407 discovery previously known as all or default

rusefillc, 2023-02-16:
TS unable to download from https://rusefi.com/online/ini/? Looks because we have not uploaded anything?

[image]

[image]

[image]

[image]

rusefillc, 2023-02-16:
@chuckwagoncomputing can you help?

closed 2023-02-17 by rusefillc

chuckwagoncomputing, 2023-02-17:
I apologize for the oversight. I thought https://github.com/rusefi/rusefi/blob/4d52a48efee6c984a3db9bceff1eb909a984ef94/firmware/gen_config_board.sh#L20-L24 check meant INI parameter wasn't needed. I'll look into it when I get a chance.

chuckwagoncomputing, 2023-02-17:
Oh I see, ini-file is used in another step.