rusefi/rusefi #5092: Something should be renamed to "4G63 No VVT"

rusefillc, 2023-02-16:
@ElDominio please spell out what exactly should be renamed to "4G63 No VVT" just to reduce confusion and I will be happy to rename right away!

mck1117, 2023-02-16:
we should just remove TT_MITSU_BOTH_BOTH

mck1117, 2023-02-16:
aka "4g93 both fronts"

ElDominio, 2023-02-17:
Because there is a 4G63 WITH VVT, and a 4G63 WITHOUT VVT

The trigger we are working on DOES NOT HAVE VVT, and now it has to be picked as a VVT type trigger
[image]

mck1117, 2023-02-17:
rusEFI (maybe erroneously) uses the word VVT to refer to "independent cam input", which is the case here

as far as the cam input side of things is concerned, this engine "is VVT" but can't actually move the cam

ElDominio, 2023-02-17:
Then yes,

4G63 exists in both VVT and non-VVT variants, and the Cam trigger shape is different between them, so they should have different names to be ready for possible future installation in said VVT 4G63

mck1117, 2023-02-17:
> Cam trigger shape is different between them

wait, really?

ElDominio, 2023-02-17:
Actually, rename to "4G93"

I have stated I'm working on a 4G93, idk why it is assumed it's the same as 4G63

mck1117, 2023-02-17:
I'm working with eldominio to do a cleanup here

rusefillc, 2023-04-02:
@ElDominio since sooooo much has changed? please create new issue if any change desired on current snapshot