rusefi/rusefi #5097: proteus F7 shows zero MSD in Windows 10

rusefillc, 2023-02-18:
Proteus F7
* does not show virtual "help" disk is this bug or feature for F7 specifically?
* does not show SD card MSD

rusefillc, 2023-02-18:
@nmschulte @mck1117 can you please test on your end? is this something with my desktops or boards, or is this a firmware issue?

nmschulte, 2023-02-18:
I'll try and test this evening or tomorrow morning.

mck1117, 2023-02-18:
SD card should work. It worked for me yesterday. Please check what you see in device manager? Old st driver instead of built in windows driver?

MSD should NOT work, see comment in efi features disabling it because 512k F7

rusefillc, 2023-02-19:
* one Proteus F4 running snapshot shows SD card on Windows 11 upstairs

closed 2023-02-19 by rusefillc

mck1117, 2023-02-19:
On MM, what do SD indicators in TS show?

rusefillc, 2023-02-19:
@mck1117 see https://github.com/rusefi/alphax-2chan/issues/84