rusefi/rusefi #5098: automated solution for daily releases?

rusefillc, 2023-02-18:
problem statement: @ElDominio approaches me asking for Jan 9, 2023 build

@chuckwagoncomputing any "cheap" way for daily release? open question where do we upload considering 1Gb for all releases (is it 1gb?

nmschulte, 2023-02-18:
I have a system we can use for storage (2TB: plenty) and have some ideas about integrating it with rusefi.com.  I'll work to get it operational this evening and touch base.

chuckwagoncomputing, 2023-02-18:
Workflows can be run on a schedule.
We could either run build-firmware on a schedule as well, and upload to different RUSEFI_SSH_SERVER if it was triggered by the schedule instead of by push etc. - or we could make a copy of build-firmware and have two different workflows.
I would prefer the first, as it's less to maintain. I think we would only need to touch the "Configs Set SSH variables" step.

If they are 1GB, it could run over 5 years before it filled 2TB.
@nmschulte could run a local cron job to remove builds older than <x> to keep his disk from filling, or we could add another workflow that would SSH into his server and do it.

rusefillc, 2023-02-19:
@mi-hol can you help by reading github documentation? do they limit storage usage or can we simply programmatically make a githib pre-release daily, would github allow that from storage cost perspective?

@mck1117 how much does 1GB cost -microsoft- github?

mi-hol, 2023-02-19:
> github documentation - do they limit storage usage

https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases#storage-and-bandwidth-quotas says:
"Each file included in a release must be under 2 GB. There is **no limit on the total size of a release, nor bandwidth usage**."

=> daily release should be possible

> how much does 1GB cost -microsoft- github?

As I understand it, you can't buy GitHub space for free tier

mi-hol, 2023-02-19:
on the topic [Repository size limits](https://docs.github.com/en/repositories/working-with-files/managing-large-files/about-large-files-on-github#repository-size-limits)

"We recommend repositories remain small, ideally less than 1 GB, and less than 5 GB is strongly recommended. Smaller repositories are faster to clone and easier to work with and maintain. If your repository excessively impacts our infrastructure, **you might receive an email from GitHub Support asking you to take corrective action**.
...
You can prevent your repository from impacting our infrastructure by effectively managing your repository's size and overall health. You can find advice and a tool for repository analysis in the [github/git-sizer](https://github.com/github/git-sizer) repository."

PS C:\Users\> git-sizer.exe

1. Analysis for rusefi/rusefi_documentation:
```
Processing blobs: 7543
Processing trees: 7598
Processing commits: 5329
Matching commits to trees: 5329
Processing annotated tags: 0
Processing references: 6
| Name                         | Value     | Level of concern               |
| ---------------------------- | --------- | ------------------------------ |
| Biggest objects              |           |                                |
| * Blobs                      |           |                                |
|   * Maximum size         [1] |  93.6 MiB | *********                      |
|                              |           |                                |
| Biggest checkouts            |           |                                |
| * Maximum path length    [2] |   174 B   | *                              |
| * Total size of files    [3] |  3.15 GiB | ***                            |

[1]  e19bb35c998c544950e9e4698f357c4e195f617b (refs/heads/**master:OEM-Docs/Renault/logan_electric_cd.iso**)
[2]  52d31fece25054f766f3044655430c95b38d806b (refs/heads/master^{tree})
[3]  157995e620cb06178d7691300fd3d181a93afc61 (05b72f9fd79a70a4e41948261d8884a724578863^{tree})
```

2. Analysis for rusefi/rusefi:
```

Processing blobs: 170368
Processing trees: 176568
Processing commits: 22519
Matching commits to trees: 22519
Processing annotated tags: 10
Processing references: 81
| Name                         | Value     | Level of concern               |
| ---------------------------- | --------- | ------------------------------ |
| Overall repository size      |           |                                |
| * Blobs                      |           |                                |
|   * Total size               |  36.8 GiB | ***                            |
|                              |           |                                |
| Biggest objects              |           |                                |
| * Blobs                      |           |                                |
|   * Maximum size         [1] |  19.9 MiB | **                             |
|                              |           |                                |
| Biggest checkouts            |           |                                |
| * Maximum path depth     [2] |    15     | *                              |
| * Maximum path length    [3] |   154 B   | *                              |

[1]  8f133523e896919abfe021433808032becf6a63c (5f1fba0eb3b2084c9bcbea5e6564138009b43993:**misc/install/TS/TunerStudioMS_Beta_v2.6.0.exe)**
[2]  c6b11e6c2ef3f558ad1a05e3524b40063968d48d (ef7440a4ba0e43f941b711e9c1da85ddec2c925b^{tree})
[3]  1d7dc81ff8db62c4f3744d800e5d34c1b7411d98 (0c70061efb98139b1addd2d596f9fc633101ba83^{tree})

```

rusefillc, 2023-02-19:
@mi-hol great info! But releases files are probably separately and probably do not count against it?

36gb of history wow. Whatever we want to discuss should be separate ticket

nmschulte, 2023-02-19:
I couldn't find anything about releases counting against repo size, but there used to be a "soft limit" of 100 GB repo size max.
I too am surprised at the 36 GiB of history!

rusefillc, 2023-02-19:
https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository

chuckwagoncomputing, 2023-02-24:
#5127 now creates a Github release every day.

closed 2023-02-24 by chuckwagoncomputing