rusefi/rusefi #5100: Add an ETB Master "On/Off" switch

ElDominio, 2023-02-18:
There should be a master On/Off switch that changes all configurable ETB-related settings to be either active or inactive. Users complain about having to switch about 4 dropdown menus to "NONE" to get an ECU that doesn't complain about ETB settings being correct when operating a cable throttle

mck1117, 2023-02-19:
what combo of default settings complain? Link a tune that complains that you think shouldn’t?

ElDominio, 2023-02-19:
People having to navigate to "TPS2" under TPS, "PPS1" "PPS2" under accel pedal, setting them to NONE, and going to ETB and setting both H bridges to false.

Getting the tune is hard since it's a reseller, but he wired TPS to TPS1 on a cable throttle and it would error out (would only be correct in idle and WOT) since TPS2 didn't move. I had to guide him through turning all those things off

mck1117, 2023-02-19:
> "TPS2" under TPS, "PPS1" "PPS2" under accel pedal, setting them to NONE

but those are all none by default?

mck1117, 2023-02-19:
I'm trying to understand how you end up stuck in this state in the first place

ElDominio, 2023-02-19:
They are all populated by default

mck1117, 2023-02-19:
what board?

ElDominio, 2023-02-19:
4chan

mck1117, 2023-02-20:
> 4chan

4chan doesn't have those pins set by default. I don't understand what the request is.

ElDominio, 2024-02-23:
idk what i was ccrying about at this point