rusefi/rusefi #5104: "Build only one bundle fast" hack mode of operation

rusefillc, 2023-02-19:
We have so much amazing automation like code generation, ini upload etc that I am addicted to github actions now.

With 20 targets three commits would create a nasty queue of builds

We can either fix the root cause of agents shortage by throwing CPU at the problem or shall we implement a shortcut:

So when I am in the field and I need to fix stuff again and again I need to only build say 4chan. At the moment I do nasty https://github.com/rusefi/rusefi/commit/47a59294058480a67e26b0a12feab65ab96d1a69

Could be that it's best as is. Or could it be improved somehow? Looking for ideas.

chuckwagoncomputing, 2023-02-20:
I could add functionality to check the commit message for something like `only:alphax-4chan`

rusefillc, 2023-02-20:
YES YES YES YES

@chuckwagoncomputing you are a genius that's an amazing idea!

rusefillc, 2023-02-21:
@chuckwagoncomputing i think it does not magic yet see https://github.com/rusefi/rusefi/commit/6e96929d4dc564729730cbea0972e5270a10c46b for instance

chuckwagoncomputing, 2023-02-21:
To use this mode, put `only:<bundle name>` in your commit message.
There must not be a space between `only:` and the name, and the name is case-sensitive.

rusefillc, 2023-02-21:
can I do
```
hello on top

only:<bundle name>
```

or at least
```
only:<bundle name>

hello on the bottom
``

or does it have to be single line only bundle? is it trivial to allow at least  the ``hello on the bottom`` option?

chuckwagoncomputing, 2023-02-21:
It should work with it anywhere in the commit message, multiple lines should be no problem (anymore).

rusefillc, 2023-02-21:
trying https://github.com/rusefi/rusefi/commit/9a5b2b1fd3892b2393a22483fd20f4c3b1d75792

rusefillc, 2023-02-21:
looks great 

thank you for your efforts @chuckwagoncomputing

rusefillc, 2023-03-01:
@chuckwagoncomputing do we expect https://github.com/rusefi/rusefi/commit/d7778b258d018e0bff7956e6bc1c76d1876f24a5 to be a case of partial buld?

PR #5115 (2023-02-21): Build only one bundle
To use this mode, put `only:<bundle name>` in your commit message.
There must not be a space between `only:` and the name, and the name is case-sensitive.
Fix #5104
PR #5148 (2023-03-02): Remove stray space
This should fix #5104 
I'm not sure how this happened, I think I re-typed the command at some point from memory without actually understanding what the -d parameter did.
closed 2023-03-02 by rusefillc