rusefi/rusefi #5106: Duplicated pin names

chuckwagoncomputing, 2023-02-19:
Duplicated pin names in board mappings now result in a failed interactive pinout generation workflow.
There are currently 5 boards with this issue.

|yaml|pins|
|--------|-------|
|hellen/hellen-gm-e67/connectors/main.yaml|J4_4|
|hellen/hellen121nissan/connectors/main.yaml|51a, 77a|
|hellen/hellen154hyundai/connectors/main.yaml|K65|
|hellen/hellen64_miataNA6_94/connectors/main.yaml|1G, 3I|
|hellen/hellen72/connectors/main.yaml|4I|

rusefillc, 2023-02-19:
@chuckwagoncomputing somehow I was not able to located miata duplicates but just pushed top three files at least

chuckwagoncomputing, 2023-02-19:
I manually confirmed the last 2, the duplicates are right above each other.

rusefillc, 2023-02-19:
@chuckwagoncomputing my bad, sorry

commit 7cf59ac0 (2023-02-19): Duplicated pin names fix #5106
commit 00a10bfe (2023-02-19): Duplicated pin names fix #5106
closed 2023-02-19 by rusefillc