rusefi/rusefi #5110: Explicit comment to un-assign second channel

rusefillc, 2023-02-20:
When second channel is not needed and not selected it's not visible - everything is easy

We have to enable second channel when it's not needed but selected in order to allow user to un-select it

Well, technically should we simply programmatically un-select it? Can we?
Well, technically we can initialize physical pin based on trigger wheel shape but we do not do that _yet_.

As a shortcut, while we are looking for ways to programmatically do better at least we can tell user to REMOVE value.

rusefillc, 2023-02-20:
[image]

mck1117, 2023-02-20:
we know whether the trigger is going to use that input or not, why can't we just ignore it in software if we don't need the second channel?

rusefillc, 2023-02-20:
> Well, technically we can initialize physical pin based on trigger wheel shape but we do not do that _yet_.

Yes we can. I suggest separate ticket for that improved solution.

commit 75eba3b9 (2023-02-20): Explicit comment to un-assign second channel fix #5110