rusefi/rusefi #5111: Feb20: 3rd ticket for 4g9x mess

rusefillc, 2023-02-20:
@ElDominio says snapshot does not start 4g9x while Jan 9th was starting 4g9x

ElDominio, 2023-02-20:
[image]

car runs, not sure if timing on EVERY cylinder is ok, but on #1, it does run

[Uploading 2023-02-20_17.10.01.mlg.zip…]()

ElDominio, 2023-02-20:
[image]

[4g93 version 11 log.zip](https://github.com/rusefi/rusefi/files/10787839/4g93.version.11.log.zip)

coils look crazy, did not attempt to start

ElDominio, 2023-02-20:
[image]

[4g93 version 29.zip](https://github.com/rusefi/rusefi/files/10787924/4g93.version.29.zip)

also starts and runs; doesn't seem too different from mess 4

ElDominio, 2023-02-20:
[image]

[4g93 version 34.zip](https://github.com/rusefi/rusefi/files/10787934/4g93.version.34.zip)

also doesn't seem to different from mess 4 and 29, all have cylinder 1 at the same angle with timing light

rusefillc, 2023-02-20:
trigger 11 gives us "CUSTOM_PRIMARY_TOO_MANY_TEETH = 9002" warning

rusefillc, 2023-02-20:
trigger 11 coil schedule does not look healthy, instant RPM is insane

[image]

ElDominio, 2023-02-20:
[4g93 v11 logs.zip](https://github.com/rusefi/rusefi/files/10788073/4g93.v11.logs.zip)

logs in different modes in v11; 
cmp f means SECONDARY is FALSE, cmp t means SECONDARY is TRUE
ckp f means PRIMARY is FALSE, ckp t means PRIMARY is TRUE

ElDominio, 2023-02-20:
[image]

[custom tooth with vvt cam inverted cam true.zip](https://github.com/rusefi/rusefi/files/10788146/custom.tooth.with.vvt.cam.inverted.cam.true.zip)

VVT seems to change, angle so does ignition angle until VVT angle stops jumping around

ElDominio, 2023-02-20:
[image]

[custom tooth with vvt cam inverted cam flse.zip](https://github.com/rusefi/rusefi/files/10788150/custom.tooth.with.vvt.cam.inverted.cam.flse.zip)

same deal

rusefillc, 2023-02-20:
> [custom tooth with vvt cam inverted cam true.zip](https://github.com/rusefi/rusefi/files/10788146/custom.tooth.with.vvt.cam.inverted.cam.true.zip)
> 
> VVT seems to change, angle so does ignition angle until VVT angle stops jumping around

not amazing with weird start and double RPM

[image]

rusefillc, 2023-02-20:
two cam true I would expect primary rise/fall to behave same and trigger gap pattern to be same

[image]

[image]

ElDominio, 2023-02-20:
[redone v11 logs with power cycle between changes.zip](https://github.com/rusefi/rusefi/files/10788401/redone.v11.logs.with.power.cycle.between.changes.zip)

rusefillc, 2023-02-21:
@mck1117 see two "cam=false" https://github.com/rusefi/rusefi/issues/5111#issuecomment-1437674127 logs above

I am at the end of my rope so I am wondering if magic constant 90 degrees in instantRpm calculator is messing with us here. Let's see if 135 would better https://github.com/rusefi/rusefi/commit/61eefa205b2034598f2bd31ef28f7934bcaffac5

rusefillc, 2023-02-21:
[image]

new feature - trigger shape into regular  @ElDominio please make a new cranking log at highest data rate TS setting

[image]

ElDominio, 2023-02-21:
[4g93 v11 on feb21 with triggerChannels view.zip](https://github.com/rusefi/rusefi/files/10796417/4g93.v11.on.feb21.with.triggerChannels.view.zip)

log with newest snapshot in cranking

ElDominio, 2023-02-21:
[image]

Overlay of https://github.com/rusefi/rusefi/issues/5111#issuecomment-1438907081 and this self stim log 

[selfstim.zip](https://github.com/rusefi/rusefi/files/10796871/selfstim.zip)

Green circles indicate places where stim vs reality align perfectly. The expected one vs reality seems spot on, why is ignition schedule crazy?

ElDominio, 2023-02-21:
[trg75 cranking log both CRANK and CAM false C7 CAM set to primary.zip](https://github.com/rusefi/rusefi/files/10798742/trg75.cranking.log.both.CRANK.and.CAM.false.C7.CAM.set.to.primary.zip)

using trg75

rusefillc, 2023-02-22:
[image]

rusefillc, 2023-02-22:
@ElDominio see how we have have vvtSyncGapRatio and then we do not have it @mck1117 any ideas why would we not have vvtSyncGapRatio any longer?

[image]

rusefillc, 2023-02-22:
i think it's time to start moving vvt position, i think we break engine phase when crank happens unexpectedly

[image]

ElDominio, 2023-02-22:
[vvt angle logs plus minus 60.zip](https://github.com/rusefi/rusefi/files/10799086/vvt.angle.logs.plus.minus.60.zip)

ElDominio, 2023-02-22:
[ignition enabled vvt trigger.zip](https://github.com/rusefi/rusefi/files/10799123/ignition.enabled.vvt.trigger.zip)

using custom tooth 2 no missing, vvt mode 4g93

rusefillc, 2023-02-22:
> [vvt angle logs plus minus 60.zip](https://github.com/rusefi/rusefi/files/10799086/vvt.angle.logs.plus.minus.60.zip)

WOW both show happy engine phase PROGRESS???

minus 60

[image]

plus 60

[image]

rusefillc, 2023-02-22:
> [ignition enabled vvt trigger.zip](https://github.com/rusefi/rusefi/files/10799123/ignition.enabled.vvt.trigger.zip)

again we are back crazy engine phase. is it just random? :(

[image]

ElDominio, 2023-02-22:
[vvt logs plus minus 60 vvt mode 4g93.zip](https://github.com/rusefi/rusefi/files/10799170/vvt.logs.plus.minus.60.vvt.mode.4g93.zip)

mck1117, 2023-02-22:
there's an extra tooth here that threw off sync until just after the cursor

[image]

rusefillc, 2023-02-22:
Yes, that's a smoking gun.

mck1117, 2023-02-22:
extra blip is perfectly aligned with coil 1 firing, smells like hardware to me

[image]

mck1117, 2023-02-22:
and once it re-syncs after that blip, firing order looks correct:

[image]

rusefillc, 2023-02-22:
https://github.com/rusefi/alphax-4chan/issues/87 would be simplest option of alternative hardware

ElDominio, 2023-02-25:
I know I'll get pushback for this but here I go

Can't we do something like, confirm CAM angle is consistent for 3 cycles then lock the VVT angle? That would work even if there's noise on startup (since it seems the noise only happens the first few cycles)

ElDominio, 2023-02-25:
I just did a log to check

I can see the VVT angle is either -20, or -200, sometimes it moves by 180 degrees.

Attached are said log and its matching tune

[5 starts with different angles.zip](https://github.com/rusefi/rusefi/files/10832450/5.starts.with.different.angles.zip)

rusefillc, 2023-02-26:
Technically yes, it looks like it's the spark noise specifically which causes noise thus technically some smarter software in fact might mitigate the issue.

But that's hella expensive in terms of software, testing and forever support. While we still have a hope for HW progress I would prefer to give that hardware approach a try. I hope to ship you a board with hacks tomorrow https://github.com/rusefi/alphax-4chan/issues/102

ElDominio, 2023-03-02:
https://rusefi.com/online/view.php?log=1178

Log with new HW mod 
tune is wadjacent to log on rusefi online

ElDominio, 2023-03-02:
second log, ignition on https://rusefi.com/online/view.php?log=1179

rusefillc, 2023-03-02:
might be OK

[image]

mck1117, 2023-03-02:
crank looks clean, still maybe some vvt channel noise

[image]

ElDominio, 2023-03-02:
[changing angle incorrect orders.zip](https://github.com/rusefi/rusefi/files/10876546/changing.angle.incorrect.orders.zip)

sometimes, changing the trigger angle by 5 degrees makes the timing mark completley vanish

there seems to be a magic point in trigger angle where it goes completley out of line
titles of logs are descriptive

ElDominio, 2023-03-03:
CAR TURNED ON

Log with matching tune: https://rusefi.com/online/view.php?log=1180

Still problems during cranking
https://rusefi.com/online/view.php?log=1181

rusefillc, 2023-03-04:
first I suspected bad cam signal on top but then I say battery dips?

[image]

ElDominio, 2023-03-04:
i think battery dips are when the engine backfires because of incorrect ignition timing while starting; since the engine gets "stuck" and the starter is in pain

ElDominio, 2023-03-07:
I expected jan9th to run on cam only, not both wheels. Would like to see if what I tested was real or just my imagination. Very close to grinding one tooth off and running as 2+1 trigger.

rusefillc, 2023-03-07:
@ElDominio i do not get the plan but here are the binaries

https://rusefi.com/temp/jan09-strange-approach/

ElDominio, 2023-03-07:
https://rusefi.com/online/view.php?log=1189

Log with most recent snapshot

https://rusefi.com/online/view.php?log=1190

also log with most recent snapshot

ElDominio, 2023-03-07:
So i have confirmed this cam wheel is same as late gen 4g63, which means i have another car to test on; the different trigger is 4G94 (so, 4G63 and 4G93 use the SAME trigger)

The pattern on 4G63/4G93 SHOULD be the same as Miata NA, but divided in Cam/Crank (phyiscally apart)

We ground off one tooth to use Single Tooth, but it still has sync issues.

https://rusefi.com/online/view.php?log=1191

ElDominio, 2023-03-07:
[image]

ElDominio, 2023-03-07:
[image]

ElDominio, 2023-03-07:
[image]

ElDominio, 2023-03-07:
[image]
 
Camcounter increased even if there is no cam signal in that point?

rusefillc, 2023-03-08:
@ElDominio was it C3+C7 previously? Is it still C3+C7 or has something changed? Any additional info on pinout situation?

rusefillc, 2023-03-08:
It's very hard for me. Looks like problem statement is slowly shifting and I am very very lost.

At this point it seems reasonable to assume hardware issues. Let's make an adapter for Proteus to be directly in place of 4chan using 4chan harness.

Reminder that @ElDominio has three sets of 8chan and three sets of Proteus plugs exactly for this purpose.

ElDominio, 2023-03-24:
new logs with modified 4chan
[DataLogs.zip](https://github.com/rusefi/rusefi/files/11067184/DataLogs.zip)

rusefillc, 2023-03-24:
@ElDominio need executive summary please. Same? Good? Bad?

rusefillc, 2023-03-24:
@ElDominio https://github.com/rusefi/alphax-4chan/issues/104

[image]

ElDominio, 2023-03-25:
Behavior on one of the modified ecus is the same as unmodified

I have noticed that it changes by 540 degrees (180*3)
[image]

[image]

[image]

rusefillc, 2023-03-26:
@ElDominio is it easy to test with the other board?

bad news this one says C200 on the sticker i.e. this one is the "more" modified board :(

ElDominio, 2023-03-26:
I'll try to get the ecu tomorrow to test

ElDominio, 2023-03-28:
Got ecu, will test tomorrow

ElDominio, 2023-03-29:
Other modded ECU (already tested one version of mod)
4 starts in log, final start has same angle problem as previous ECU (540 degrees apart)

[image]
Since I am getting this error, here is the current tune in a zip file
[CurrentTune.zip](https://github.com/rusefi/rusefi/files/11104369/CurrentTune.zip)

And here is log of current ECU
[othermoddedecucrankinglog.zip](https://github.com/rusefi/rusefi/files/11104373/othermoddedecucrankinglog.zip)

rusefillc, 2023-03-29:
and unmodified we have somewhere?

[5 more replies]

closed 2023-03-31 by rusefillc