rusefi/rusefi #5119: Custom trigger dialog? What adjustments do we like most?

rusefillc, 2023-02-21:
pic from https://www.facebook.com/groups/rusEfi

[image]

mck1117, 2023-02-21:
that's a completely different ini (and looks like it's pretty old?), we've never had a menu called "E-Throttle"

mck1117, 2023-02-21:
hello friend 
[image]

rusefillc, 2023-02-21:
Hello @960

mck1117, 2023-02-22:
this is basically what I did in #5077

rusefillc, 2023-02-22:
@ElDominio opinions?

ElDominio, 2023-02-22:
Uuuh the current iteration i like very much, is more change necessary?

mck1117, 2023-02-22:
> Uuuh the current iteration i like very much, is more change necessary?

Good answer!

closed 2023-02-22 by mck1117