rusefi/rusefi #5122: Caching dependencies to speed up workflows

rusefillc, 2023-02-22:
https://docs.github.com/en/actions/using-workflows/caching-dependencies-to-speed-up-workflows

nmschulte, 2023-02-22:
can cache `/var/cache/apt` and the GCC ARM EABI download, and probably a bit more

chuckwagoncomputing, 2023-02-22:
Note that the cache is on a Github server, not actually on the runner.
Self-hosted runners don't benefit from this at all.
I'm not sure how much Github-hosted runners would benefit, but I assume it would be a noticeable amount.

Here's what this would require in terms of workflow steps:
1. A step to check if the file(s) are already downloaded, if we want to support self hosted runners (I do)
2. The cache step
3. A step to download the files if they weren't in cache. This already exists for gcc-arm-eabi.

My personal opinion is that the benefit doesn't outweigh the added complexity.