rusefi/rusefi #5123: Why does rusEFI devour COM ports?

ElDominio, 2023-02-23:
Every new rusEFI I connect to my laptop creates a new COM port, while most other ECU's use the same COM port assigned to the USB hardware they use (all of the same brand use the same COM port)

Having two rusEFI's connected at one is an edge case, but can't they be made to at least not request a new COM port every time?

rusefillc, 2023-02-24:
@dron0gus meaning that if we want to we can give boards per-board unique IDs can we @mck1117 ?

dron0gus, 2023-02-24:
We already have per-board unique ID that is produced from per-MCU unique SN. See https://github.com/rusefi/rusefi/blob/master/firmware/hw_layer/ports/stm32/serial_over_usb/usbconsole.cpp#L24

rusefillc, 2023-02-24:
And I mean we can change it from per unique MCU make it per unique board design, more or less return hellen board rev id or something similar

dron0gus, 2023-02-25:
You can report board type and its revision through Device Descriptor string. Now it is "rusEFI Engine Management ECU" for all boards.

ElDominio, 2023-02-25:
While I understand that having an eternal COM port list is not really a problem for TunerStudio/rusEFI, a lot of older engine management (and heck, some newer) programs start complaining about COM ports, or plainly don't have a list that goes up past 10

[image]

If anyone could point me to an app that flushes my COM ports so I don't have to remove one by one it would be great

mck1117, 2023-02-25:
> change it from per unique MCU make it per unique board design

hardware ci relies on stable serial numbers for exact chip

mck1117, 2023-02-25:
> If anyone could point me to an app that flushes my COM ports so I don't have to remove one by one it would be great

Not that I know of, no. You could probably write a powershell script to do it if motivated enough.

rusefillc, 2023-02-26:
> hardware ci relies on stable serial numbers for exact chip

hardware ci only _needs_ to distinguish between kinds of boards not between specific board pieces?

esaulenka, 2023-03-24:
> You could probably write a powershell script to do it if motivated enough.

@mck1117, thanks for good idea.
Never tried to do so, but it looks quite easy: https://stackoverflow.com/questions/60638900/uninstall-device-from-powershell

ElDominio, 2024-02-23:
This is still reasonable. Maybe include an applet to flush COM ports inside of console? idk

ElDominio, 2024-05-25:
is this important in any way