rusefi/rusefi #5134: we have weird unconditional coupling with ST_STM32F4_DISCOVERY

rusefillc, 2023-02-26:
at the moment https://github.com/rusefi/rusefi/blob/master/firmware/hw_layer/ports/stm32/stm32f4/hw_ports.mk says

```
ALLCSRC += $(CHIBIOS)/os/hal/boards/ST_STM32F4_DISCOVERY/board.c
```

which is pretty insane

same drama for F7 H7 etc

mck1117, 2023-02-26:
why is that insane?

look at that file, it's completely generic

rusefillc, 2023-02-26:
https://github.com/rusefi/ChibiOS/commit/c6b439295b5fd47eaf6332ea224c6b476d570fb8
https://github.com/rusefi/ChibiOS/commit/fd3770d71d6b9dae6185ff7ad32746e88a0a8f1d
https://github.com/rusefi/ChibiOS/commit/284be36d608557fb6f9c58786d9da1d6a9283a9c

is it in fact completely generic see above? @dron0gus wonders if random AF on random pins is a good thing for https://github.com/andreika-git/hellen-one/issues/262

mck1117, 2023-02-26:
the `*.c` is generic, the `*.h` is not necessarily generic.

mck1117, 2023-02-26:
Also, the .h edited in those commits is not actually being used - the one being used is at `firmware/hw_layer/ports/stm32/stm32f4/board.h`

Confirmed this by putting some non-compileable trash in the chibios one and the firmware still builds.

rusefillc, 2023-02-26:
Ohhh

I shall push a comment at least

dron0gus, 2023-02-26:
Oh, it is even more confusing...

rusefillc, 2023-02-27:
reverted chibi branch, added comment

closed 2023-02-27 by rusefillc