rusefi/rusefi #5136: SD card drive does not work on specific Win11 device

rusefillc, 2023-02-27:
latest 2chan finally has SD card working!

On Win10 it works as expected
On my primary Win11 laptop it does not work while Device Manager even shows E:\ :(

[image]
[image]

rusefillc, 2023-03-23:
same or not? https://rusefi.com/forum/viewtopic.php?p=47443#p47443

rusefillc, 2023-05-07:
happy windows 10 with 4chan rev F
```
Information for device rusEFI Engine Management ECU (VID=0x0483 PID=0x5740):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x000E
Current configuration value: 0x01
Number of open pipes: 5

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0xEF	bDeviceClass      (Miscellaneous device)
0x02	bDeviceSubClass   
0x01	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x0483	idVendor
0x5740	idProduct
0x0200	bcdDevice
0x01	iManufacturer   "rusEFI LLC"
0x02	iProduct        "rusEFI Engine Management ECU"
0x03	iSerialNumber   "070017000847323134353232"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0062	wTotalLength   (98 bytes)
0x03	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0xC8	bMaxPower      (400 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x08	bInterfaceClass      (Mass Storage Device Class)
0x06	bInterfaceSubClass   (Transparent SCSI subclass)
0x50	bInterfaceProtocol   (Bulk only transport)
0x00	iInterface

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x01	bEndpointAddress  (OUT endpoint 1)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Interface Association Descriptor:
------------------------------
0x08	bLength
0x0B	bDescriptorType
0x01	bFirstInterface
0x02	bInterfaceCount
0x02	bFunctionClass      (Communication Device Class)
0x02	bFunctionSubClass   (Abstract Control Model - ACM)
0x01	bFunctionProtocol   (ITU-T V.250)
0x02	iFunction   "rusEFI Engine Management ECU"

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x02	bInterfaceClass      (Communication Device Class)
0x02	bInterfaceSubClass   (Abstract Control Model - ACM)
0x01	bInterfaceProtocol   (ITU-T V.250)
0x00	iInterface

CDC Header Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x00	bDescriptorSubtype
0x0110	bcdCDC

CDC Call Management Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x01	bDescriptorSubtype
0x00	bmCapabilities
0x02	bDataInterface

CDC Abstract Control Management Functional Descriptor:
------------------------------
0x04	bFunctionalLength
0x24	bDescriptorType
0x02	bDescriptorSubtype
0x02	bmCapabilities

CDC Union Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x06	bDescriptorSubtype
0x01	bControlInterface
0x02	bSubordinateInterface(0)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x83	bEndpointAddress  (IN endpoint 3)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0xFF	bInterval         (255 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x02	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x0A	bInterfaceClass      (CDC Data)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "rusEFI LLC"
0x02   0x0409  "rusEFI Engine Management ECU"
0x03   0x0409  "070017000847323134353232"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
rusEFI Engine Management ECU (VID=0x0483 PID=0x5740) Port: 7

Running on: Windows 10 or greater (Build Version 19045)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

```

rusefillc, 2023-05-07:
windows 11 happy with same 4chan rev F unexpected!

```
Information for device rusEFI Engine Management ECU (VID=0x0483 PID=0x5740):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0008
Current configuration value: 0x01
Number of open pipes: 5

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0xEF	bDeviceClass      (Miscellaneous device)
0x02	bDeviceSubClass   
0x01	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x0483	idVendor
0x5740	idProduct
0x0200	bcdDevice
0x01	iManufacturer   "rusEFI LLC"
0x02	iProduct        "rusEFI Engine Management ECU"
0x03	iSerialNumber   "070017000847323134353232"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0062	wTotalLength   (98 bytes)
0x03	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0xC8	bMaxPower      (400 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x08	bInterfaceClass      (Mass Storage Device Class)
0x06	bInterfaceSubClass   (Transparent SCSI subclass)
0x50	bInterfaceProtocol   (Bulk only transport)
0x00	iInterface

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x01	bEndpointAddress  (OUT endpoint 1)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Interface Association Descriptor:
------------------------------
0x08	bLength
0x0B	bDescriptorType
0x01	bFirstInterface
0x02	bInterfaceCount
0x02	bFunctionClass      (Communication Device Class)
0x02	bFunctionSubClass   (Abstract Control Model - ACM)
0x01	bFunctionProtocol   (ITU-T V.250)
0x02	iFunction   "rusEFI Engine Management ECU"

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x02	bInterfaceClass      (Communication Device Class)
0x02	bInterfaceSubClass   (Abstract Control Model - ACM)
0x01	bInterfaceProtocol   (ITU-T V.250)
0x00	iInterface

CDC Header Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x00	bDescriptorSubtype
0x0110	bcdCDC

CDC Call Management Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x01	bDescriptorSubtype
0x00	bmCapabilities
0x02	bDataInterface

CDC Abstract Control Management Functional Descriptor:
------------------------------
0x04	bFunctionalLength
0x24	bDescriptorType
0x02	bDescriptorSubtype
0x02	bmCapabilities

CDC Union Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x06	bDescriptorSubtype
0x01	bControlInterface
0x02	bSubordinateInterface(0)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x83	bEndpointAddress  (IN endpoint 3)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0008	wMaxPacketSize    (1 x 8 bytes)
0xFF	bInterval         (255 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x02	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x0A	bInterfaceClass      (CDC Data)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "rusEFI LLC"
0x02   0x0409  "rusEFI Engine Management ECU"
0x03   0x0409  "070017000847323134353232"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
rusEFI Engine Management ECU (VID=0x0483 PID=0x5740) Port: 2

Running on: Windows 10 or greater (Build Version 22621)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02
```

dron0gus, 2023-05-07:
Same issue with Linux:
Bad behavior:
```
[35435.586002] usb 2-1.5: new full-speed USB device number 58 using ehci-pci
[35435.695004] usb 2-1.5: New USB device found, idVendor=0483, idProduct=df11
[35435.695006] usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[35435.695008] usb 2-1.5: Product: STM32  BOOTLOADER
[35435.695009] usb 2-1.5: Manufacturer: STMicroelectronics
[35435.695010] usb 2-1.5: SerialNumber: 3275365C3536
[35492.881549] usb 2-1.5: USB disconnect, device number 58
[35493.445924] usb 2-1.5: new full-speed USB device number 59 using ehci-pci
[35493.555827] usb 2-1.5: New USB device found, idVendor=0483, idProduct=5740
[35493.555832] usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[35493.555835] usb 2-1.5: Product: rusEFI Engine Management ECU
[35493.555837] usb 2-1.5: Manufacturer: rusEFI LLC
[35493.555839] usb 2-1.5: SerialNumber: 200042000451363532363332
[35493.556572] usb-storage 2-1.5:1.0: USB Mass Storage device detected
[35493.557134] scsi host6: usb-storage 2-1.5:1.0
[35493.557493] cdc_acm 2-1.5:1.1: ttyACM0: USB ACM device
[35494.583117] scsi 6:0:0:0: Direct-Access     rusEFI   INI Drive        v6.1 PQ: 0 ANSI: 4
[35494.584023] scsi 6:0:0:1: Direct-Access     rusEFI   SD Card          v6.1 PQ: 0 ANSI: 4
[35494.585149] sd 6:0:0:0: Attached scsi generic sg3 type 0
[35494.585935] sd 6:0:0:1: Attached scsi generic sg4 type 0
[35494.587634] sd 6:0:0:0: [sdc] Attached SCSI removable disk
[35494.589636] sd 6:0:0:1: [sdd] Attached SCSI removable disk
```
Good:
```
[32727.508914] usb 2-1.5: new full-speed USB device number 21 using ehci-pci
[32727.618840] usb 2-1.5: New USB device found, idVendor=0483, idProduct=df11
[32727.618845] usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[32727.618848] usb 2-1.5: Product: STM32  BOOTLOADER
[32727.618850] usb 2-1.5: Manufacturer: STMicroelectronics
[32727.618852] usb 2-1.5: SerialNumber: 3275365C3536
[32755.825002] usb 2-1.5: USB disconnect, device number 21
[32756.276870] usb 2-1.5: new full-speed USB device number 22 using ehci-pci
[32756.386897] usb 2-1.5: New USB device found, idVendor=0483, idProduct=5740
[32756.386903] usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[32756.386905] usb 2-1.5: Product: rusEFI Engine Management ECU
[32756.386907] usb 2-1.5: Manufacturer: rusEFI LLC
[32756.386909] usb 2-1.5: SerialNumber: 200042000451363532363332
[32756.387690] usb-storage 2-1.5:1.0: USB Mass Storage device detected
[32756.388114] scsi host6: usb-storage 2-1.5:1.0
[32756.388678] cdc_acm 2-1.5:1.1: ttyACM0: USB ACM device
[32757.397938] scsi 6:0:0:0: Direct-Access     rusEFI   INI Drive        v6.1 PQ: 0 ANSI: 4
[32757.398663] scsi 6:0:0:1: Direct-Access     rusEFI   SD Card          v6.1 PQ: 0 ANSI: 4
[32757.399729] sd 6:0:0:0: Attached scsi generic sg3 type 0
[32757.400336] sd 6:0:0:1: Attached scsi generic sg4 type 0
[32757.401533] sd 6:0:0:1: [sdd] 7698432 512-byte logical blocks: (3.94 GB/3.67 GiB)
[32757.402148] sd 6:0:0:0: [sdc] Attached SCSI removable disk
[32757.402879] sd 6:0:0:1: [sdd] Write Protect is off
[32757.402883] sd 6:0:0:1: [sdd] Mode Sense: 03 00 00 00
[32757.403715] sd 6:0:0:1: [sdd] No Caching mode page found
[32757.403724] sd 6:0:0:1: [sdd] Assuming drive cache: write through
[32757.414172]  sdd: sdd1
[32757.419192] sd 6:0:0:1: [sdd] Attached SCSI removable disk
[32759.240836] usb 2-1.5: reset full-speed USB device number 22 using ehci-pci
[32759.350127] cdc_acm 2-1.5:1.1: ttyACM0: USB ACM device
```
Bisecting shows that something was broken on eef016788fd54f8aae232ef80a4759af0e199779 or e5d199272dd3a77e308d9588513fe919d4e51c2e .
Nothing changed around USB, but `persistent_config_s` was dramatically dropped on these commits.

rusefillc, 2023-05-07:
@dron0gus @mck1117 does this smell like some buffer being sometimes in DMA-compatible RAM and sometimes into a different kind of RAM?

dron0gus, 2023-05-09:
5 sec timeout for USB enumeration? https://github.com/rusefi/rusefi/pull/5270

mi-hol, 2023-08-15:
Should this be closed via https://github.com/rusefi/rusefi/commit/cdd5161d5234da4f6f6b76e9e44055cd940abcc0 ?

closed 2023-08-15 by rusefillc