rusefi/rusefi #5145: Shall we enable '-Werror=stringop-truncation'?

rusefillc, 2023-03-01:
-Werror=stringop-truncation sounds cool but it generates at least the following error

```
./console/binary_log/log_field.cpp:47:24: error: 'strncpy' specified bound 34 equals destination size [-Werror=stringop-truncation]
   47 |                 strncpy(&buffer[55], m_category, 34);
      |                        ^
```

nmschulte, 2023-03-02:
How bad is `-Werror=all`?

nmschulte, 2023-03-02:
Does MLV/`.mlq` require that the string field here include a null terminator?  `BinarySensorLog` seems to not need it/worry: https://github.com/rusefi/rusefi/blob/master/java_console/ui/src/main/java/com/rusefi/sensor_logs/BinarySensorLog.java#L148

nmschulte, 2023-03-02:
`strncpy` behavior is the ideal solution here -- are we opposed to judicious use of `#pragma GCC diagnostic ignored`?
```cpp
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wstringop-truncation"
    // strncpy's exclusion of null terminator is correct and appropriate behavior for this field: silence the warning
    strncpy(...);
#pragma GCC diagnostic pop
```

rusefillc, 2023-03-02:
I am all for it