rusefi/rusefi #5152: weird: happy running car shows unrealistic vvtSyncGapRatio

rusefillc, 2023-03-04:
gap ratio 2000 all the time on running engine? is that some scaling issue or what?

see https://github.com/rusefi/rusefi/issues/5111#issuecomment-1454034075

[image]

rusefillc, 2023-03-07:
@ElDominio I believe that in terms of cost/benefit this is the most important issue

Please provide a log with snapshot software I am looking for data from new debug field

rusefillc, 2023-03-21:
#5193

andreika-git, 2023-03-21:
Unrealistic vvtSyncGapRatio means we have very low toothDuration sometimes, like only 8-22 ticks! This means that a reentrant protection for hwHandleVvtCamSignal() might help...

rusefillc, 2023-03-21:
> Unrealistic vvtSyncGapRatio means we have very low toothDuration sometimes, like only 8-22 ticks! This means that a reentrant protection for hwHandleVvtCamSignal() might help...

but in https://rusefi.com/online/view.php?log=1181 it's not _sometimes_ it's _always_ in a very repetitive way. The clear nice pattern and car running makes me think that the screenshot about is totally NOT about the noise

mck1117, 2023-03-21:
this is absolutely hardware noise:

[image]

there are 3 falling edges here when there should be exactly 1

mck1117, 2023-03-21:
last row red/green are number of rise/fall events that happened since the last sample, scaled so that 0 and 1 (expected values) are hidden, all spikes are duplicate edges

[image]

rusefillc, 2023-03-21:
Wow, sounds like more error validation is needed for insane vvt ratio

rusefillc, 2023-03-21:
damn I am blind :( there is an area of reasonable ratios from 0.2 to 4

[image]

mck1117, 2023-03-22:
I don't think you meant to close

rusefillc, 2023-03-22:
I did mean to close. Better error detection was implemented here, smarter noise processing out of scope, hw issues would be be hardware-specific tickets.

commit 3031818f (2023-03-21): weird: happy running car shows repetitive pattern of unrealistic vvtSyncGapRatio fix #5152
closed 2023-03-22 by rusefillc