rusefi/rusefi #5154: "different user levels in TS" idea

rusefillc, 2023-03-05:
Thank you for taking the time to report an issue!

To make our work easier, please provide as many details as you can:
* [ ] What hardware did you use while experiencing the issue?

rusefillc, 2023-03-05:
We have at least two options I can think of:
* cheaper option: a drop down "Simple/Full" or even "Setup/Tuning/Full" which would allow us to hide individual fields/panels/menues based on selected category. Pretty much "full" is current layout and we can subtract from it based on user drop down.
* expensive option: multiple completely parallel .ini files. Probably unrealistic too much effort

@ElDominio what do you think?

classicdaily, 2023-03-05:
I like the cheaper option.   Yes FULL would be how it is now.  Even Just Tuning/Full might be enough  I will think on that too.

dron0gus, 2023-03-06:
1. Installer mode - output/input assignment, triggers, etc.
2. Tuner mode - tables, curves, ets
3. Full mode.

ElDominio, 2023-03-08:
I also like the cheaper option, or what @dron0gus said

Many people's brains melt when seeing things like PID's and having those options available to most of them creates confusion.

Would the menus be completely hidden or just grayed out?

rusefillc, 2023-03-15:
could work together with https://github.com/rusefi/rusefi/issues/4420

commit 27a76a6d (2024-07-27): Full layout vs tuning layout fix #5154
closed 2024-07-27 by rusefillc