rusefi/rusefi #5159: latest snapshots cause error "firmware format incompatible" in console

mi-hol, 2023-03-08:
downloaded https://rusefi.com/build_server/rusefi_bundle_mre_f4_no_openblt.zip
gives

[image]

rusefillc, 2023-03-08:
please confirm that you have flashed rusefi_bundle_mre_f4_no_openblt.zip into hardware

As is I wonder if you have used latest console to connect to very old firmware

mi-hol, 2023-03-08:
I connected MRE via USB for first time and then pressed "connect".
Will try to flash as suggested.

mi-hol, 2023-03-08:
flashing fixed error
[image]
[image]

mi-hol, 2023-03-08:
I'll add this encountered issue as note for troubleshooting in doc

mi-hol, 2023-03-08:
@rusefillc actually a better approach would be to change the error message in code.
The recommendation to "use previous console version or flash new firmware" should be added to error message.

Benefit:
This change would allow issue and resolution to be understood by user instantly, improving user experience

mi-hol, 2023-03-08:
unfortunately searching with string "incompatible firmware format" had no hits in GitHub, so I don't know where to change it for a PR

mi-hol, 2023-03-08:
found string  now via VSCode in \rusefi\java_console\io\src\main\java\com\rusefi\binaryprotocol\BinaryProtocol.java

closed 2023-03-08 by mi-hol