rusefi/rusefi #5168: WA HW CI up monitor

rusefillc, 2023-03-10:
It looks like proteus HW CI is offline or hanging

A week ago it was discovery

How can we have computers poke @mck1117 instead of me poking @mck1117 ?

FDSoftware, 2024-11-07:
For this we can use portainer-ce since GH runners run inside docker, portainer allows you to connect to remote docker instances and manage the containers or start a console within it
https://docs.portainer.io/start/install-ce

example connecting to my own docker instance:
[image][image]

The only problem with Portainer is that it leaves out the most sporadic maintenance (updating the host OS, checking the status of the USB ports, etc.), for that I found Rundeck, I never tried it but in the documentation it shows how to create actions that run bash commands on remote servers

https://docs.rundeck.com/docs/learning/howto/ssh-on-linux-nodes.html

mck1117, 2024-11-07:
> How can we have computers poke @mck1117 instead of me poking @mck1117 ?

Those resources have been reallocated to another project.

closed 2024-11-07 by rusefillc