rusefi/rusefi #5169: Even partial build should produce proper bundle/autoupdate/upload .ini

rusefillc, 2023-03-10:
The amazing 'only:xxx' feature should behave closer to normal build, just only for one matrix entry

chuckwagoncomputing, 2023-03-10:
I'm confused about what you're trying to achieve here.
As far as I can tell, what you've done is enabled artifact upload for all pushes to master. Is that right?

rusefillc, 2023-03-10:
Before change "only honda" would create not complete autoupdate of only honda

I want "only honda" to create full autoupdate of only honda

chuckwagoncomputing, 2023-03-10:
Did your change fix it?
If so, can you explain how?

chuckwagoncomputing, 2023-03-10:
Also, can you link a run that did not work correctly?

rusefillc, 2023-03-10:
> before: https://github.com/rusefi/rusefi/actions/runs/4379928313/jobs/7666409230

problem statement: there was firmware build but autoupdate bundle was not uploaded, also .ini was not uploaded

[image]

after the change https://github.com/rusefi/rusefi/actions/runs/4380160678/jobs/7666920589 looks like it works as desired - there is both firmware and autoupdate bundle upload

chuckwagoncomputing, 2023-03-10:
That run didn't upload bundles because it wasn't a push to master, it was a push to [2023-03-10](https://github.com/rusefi/rusefi/tree/2023-03-10)

rusefillc, 2023-03-10:
We need a phone call :(

rusefillc, 2023-03-10:
Something https://github.com/rusefi/rusefi/commit/b385756ce5bb9742887d8d4d0ddd734604d7885d

rusefillc, 2023-03-10:
So funny thing https://github.com/rusefi/rusefi/commit/b385756ce5bb9742887d8d4d0ddd734604d7885d is a commit related to nightly tag and this has something to do with my problem.

rusefillc, 2023-03-10:
I was watching github action job and it looks like I was watching the wrong output?

Q1: Do I just revert my change?
Q2: Do you want to report this to GH?

chuckwagoncomputing, 2023-03-10:
Yep, I think that's what happened.

Yes, and yes.

rusefillc, 2023-03-10:
Reverted by https://github.com/rusefi/rusefi/commit/336250469946624e05720c3ea8adf967b7b512c3

closed 2023-03-10 by rusefillc

chuckwagoncomputing, 2023-03-20:
Github has responded, and says this is "by design".
Good job, Microsoft.

commit 53acf588 (2023-03-10): Even 'only' build should produce proper bundle/autoupdate/upload .ini fix #5169
only:hellen-honda-k
commit 33625046 (2023-03-10): Revert "Even 'only' build should produce proper bundle/autoupdate/upload .ini fix #5169"

This reverts commit 53acf58839cf2882be50c0f63c14bc68c4de83a3.