rusefi/rusefi #5170: Mar 9 4g results

rusefillc, 2023-03-10:
@ElDominio has news

ElDominio, 2023-03-10:
Test 1:
4chan with only one mod, which will be shipped back tomorrow, disconnected ignition coils. Signal still noisy, get double RPM sometimes

[no coils connected modded 4chan.zip](https://github.com/rusefi/rusefi/files/10937493/no.coils.connected.modded.4chan.zip)

Currently wiring test 2, 8chan. Will report back

ElDominio, 2023-03-10:
Test 2:
8chan
seems to have no noise, have started more than 10 times without noise, still have coils disconnected,

[goodcrankinglog.zip](https://github.com/rusefi/rusefi/files/10937662/goodcrankinglog.zip)

 connecting coils to test
 connected
 testing
 log:
 
[connectedcoilsccranking.zip](https://github.com/rusefi/rusefi/files/10937670/connectedcoilsccranking.zip)

going to try to turn on car

rusefillc, 2023-03-10:
Wow!

ElDominio, 2023-03-10:
> Test 2: 8chan seems to have no noise, have started more than 10 times without noise, still have coils disconnected, [...]

noticed it was running semi-sequential when i have a one tooth cam
turned on "Require Cam/VVT sync for ignition" and I never have ignition.
I can see LED pulsing correcetly

tune: https://rusefi.com/online/index.php?vehicleName=TecoTest+8chan&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

log: 
[interesting no crank sync.zip](https://github.com/rusefi/rusefi/files/10938381/interesting.no.crank.sync.zip)

*a new error has been found could not upload log to rusefi online; new ticket at 8chaan https://github.com/rusefi/alphax-8chan/issues/72

rusefillc, 2023-03-13:
drama https://github.com/rusefi/alphax-8chan/issues/81 :(

rusefillc, 2023-03-13:
@ElDominio for now please move to 5C hall 4 (with latest firmware)

https://github.com/rusefi/alphax-8chan/issues/77#issuecomment-1465427208

ElDominio, 2023-03-13:
Noted, will update asap

ElDominio, 2023-03-15:
https://rusefi.com/online/view.php?log=1199

https://rusefi.com/online/index.php?vehicleName=8chan+convert&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

ElDominio, 2023-03-15:
The timing light shows a different angle on starts where the VVT angle is not the same as last

rusefillc, 2023-04-01:
Too many tickets closing this one @ElDominio

closed 2023-04-01 by rusefillc