rusefi/rusefi #5172: huh what's wrong with proteus F4 rusefi.bin?

rusefillc, 2023-03-10:
I am confused.

I have proteus f4 powered via USB cable and three wires of https://github.com/rusefi/rusefi/wiki/Tag-Connect-TC2030 fake st-link v2

[image]

so we see verification error and proteus did not start

same exact board was happy to start once I've used stm32 DFU via only USB

rusefillc, 2023-03-11:
Just to clarify the issue is present on proper stm32 - it just came up while playing silly games.

rusefillc, 2023-03-11:
openblt would be obvious suspect but turning it off did not fix the issue https://github.com/rusefi/rusefi/commit/5e826e34a2cdb1babd649ab5c4aa04b488fe7245

@nmschulte @chuckwagoncomputing do you guys have TC2030 or the 10 pin jtag to try .bin on any proteus? try bin on any eval board with any st-link?

mck1117, 2023-03-11:
is that not the same rusefi.bin that gets flashed for HW CI?

rusefillc, 2023-04-04:
> is that not the same rusefi.bin that gets flashed for HW CI?

who knows? is it not applied using same tool? is it not same rusefi.bin?

rusefillc, 2023-04-04:
what are these three segments? are those in the room right now?

[image]

mck1117, 2023-04-04:
`08000000` is flash
`20000000` is ram

it wrote `0x9cb5c` bytes to flash, and none to ram (641k bytes or something)

rusefillc, 2023-04-05:
Hoooooold on we have https://github.com/rusefi/wideband/issues/22

I wonder if the issue here is also releted to stm32prog