rusefi/rusefi #5173: Kawasaki EX300 engine (2 cylinder, 180 degree angle), 24-2 crank issues

ElDominio, 2023-03-11:
Tune at
https://rusefi.com/online/index.php?sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

log here, rusefi online not accepting logs
[cranking4.zip](https://github.com/rusefi/rusefi/files/10947044/cranking4.zip)

This engine fires
1-2-blank-blank if you see it as a 4 cylinder, which seems to not be supported, or I'm missing some configuration?

ElDominio, 2023-03-11:
[cranking5.zip](https://github.com/rusefi/rusefi/files/10947082/cranking5.zip)

mck1117, 2023-03-11:
> 1-2-blank-blank if you see it as a 4 cylinder, which seems to not be supported, or I'm missing some configuration?

this is supported, set the offset on cylinder 2 to fire 180 degrees early with the firing order `1-2`

ElDominio, 2023-03-11:
Does not work as expected, log will be attached

I set the engine to 4cyl 1-3-4-2 on wasted spark and got it running

mck1117, 2023-03-11:
Currently odd-fire engines are not expected to run on wasted spark at all. You'll need full sync (ie, cam or map), and to set "no fire until cam sync" to true.

ElDominio, 2023-03-11:
This log runs as 4cyl wasted spark:

[running as 4cyl set to 1342 wasted spark.zip](https://github.com/rusefi/rusefi/files/10947463/running.as.4cyl.set.to.1342.wasted.spark.zip)

This log is as 2cyl wasted spark (which does not run, as you said)
[cranking with 2cyl wasted spark not running.zip](https://github.com/rusefi/rusefi/files/10947466/cranking.with.2cyl.wasted.spark.not.running.zip)

The problem is, it doesn't pull much vacuum during cranking, so i don't know if i can tell it where to sync. Are my fears unfounded?

ElDominio, 2024-02-23:
using MAP sync worked..... i think? i have two of these running with MAP sync so i think it's fine

closed 2024-02-23 by ElDominio