rusefi/rusefi #5179: What table is this? Critical error and I can't find what table it is

ElDominio, 2023-03-14:
[image]

halp

ElDominio, 2023-03-14:
[image]

rusefillc, 2023-03-14:
I think there are two things

* questionable memory layout change probably by me - root cause of fatal
* TS project does not match firmware on second image

Recommended action: Controller->popular vehicles->reset firmware and then reload from TS into controller

rusefillc, 2023-03-14:
Clearly some sort of automation is needed to guard repo from me :(

ElDominio, 2023-03-14:
> I think there are two things
> 
>     * questionable memory layout change probably by me - root cause of fatal
> 
>     * TS project does not match firmware on second image
> 
> 
> Recommended action: Controller->popular vehicles->reset firmware and then reload from TS into controller

this worked