rusefi/rusefi #5191: Coil outputs set under "Rotary" cannot be bench tested

ElDominio, 2023-03-20:
Outputs set under Rotary (so, trailing outputs) do not fire in bench test. Fire normally with engine running

mck1117, 2023-03-20:
which button?

ElDominio, 2023-03-20:
[image]

mck1117, 2023-03-21:
sorry, why would the test button for normal spark fire the trailing outputs too? isn't the point that it fires exactly one thing so you can tell that it's wired right?

rusefillc, 2023-03-21:
sounds like this is supposed to be "We have no feature to bench test trailing outputs"

[image]

ElDominio, 2023-03-21:
> sorry, why would the test button for normal spark fire the trailing outputs too? isn't the point that it fires exactly one thing so you can tell that it's wired right?

Yes

If i assign trailing to say pins 3 and 4, I can't test pins 3 and 4, only pins 1 and 2

rusefillc, 2023-03-21:
We are not very effective without a tune URL.

I can guess 2 cylinder config disabling spark 3 and 4 combined with no feature to bench test trailing as trailing

ElDominio, 2023-03-21:
https://rusefi.com/online/view.php?msq=1429

ElDominio, 2024-02-23:
So, only this week I found that rotary settings only SPECIFICALLY work with three things being true:
1) Two-stroke is chosen as  kind of stroke
2) 2 cylinder with firing order 1-2
3) Crank sensor is mounted on eccentric shaft

1 and 2 are reasonable but shoulld be highlighted in the "Rotary" section of rusEFI

Now 3, I have no idea why setting 2-stroke mode eats the location of crank trigger. Some older rotaries rotate the "crank" sensor at half speed, meaning we need cam multiplication (speccifically, this car had a 24-1 which spun at half speed, so setting 24-1 would not work in 2 stroke mode)

This specifically led me to run it as a wasted-spark 4 cylinder 4 stroke, which schedules ignition just fine; BUT
trailing spark logic goes crazy because of this. We should either critical on incompatible settings or make clear the expected setup