rusefi/rusefi #5197: 8chan Official well known idle stepper via H-bridges configuration

rusefillc, 2023-03-25:
let's use stepper IAC on 8chan

rusefillc, 2023-03-25:
See also https://github.com/rusefi/rusefi/issues/3661

rusefillc, 2023-03-25:
17123852 Throttle Position Sensor + Idle Air Control Valve 17113598 For Chevy GM

[image]

andreika-git, 2023-03-26:
We don't have enough settings to configure the h-bridge stepper with two 9201s:

[image]

We need an option to set useTwoWires = false. We already have it for ETB (etb_use_two_wires), and now we need it for the stepper. Maybe stepper_dc_use_two_wires?

rusefillc, 2023-03-27:
@andreika-git can you please add code comment with human readable comment which pins are A which pins are B?

how do I figure out where to connect "high A" where do I connect "low A" i.e. what about A/B direction?

rusefillc, 2023-03-28:
https://youtu.be/-DePdZhZXYU

https://rusefi.com/online/view.php?msq=1434

closed 2023-03-28 by rusefillc