rusefi/rusefi #5200: what is quad-wired batch injection called? do we support it?

rusefillc, 2023-03-26:
I believe we support injectors wired in pairs but do we support injectors wired in groups of 4?

rusefillc, 2023-03-27:
We have IM_BATCH
@andreika-git would we need to add IM_QUAD_BATCH?

mck1117, 2023-03-27:
what crazy setup has injectors in groups of 4? you know you can run the wrong injector timing right, just use a different mode with whatever wiring

rusefillc, 2023-03-28:
marine engines made in 2000 using 1976 technology?

rusefillc, 2023-03-28:
We like to use "proper" units right? so if that's a v8 with XXX injector flow and two groups of four how would I run it?

I would assume "proper" IM_QUAD_BATCH is around 13 lines of code total?

mck1117, 2023-03-28:
The injector flow numbers are for "injector flow per cylinder" in any mode other than single point (TBI). That means that if I set to sequential, but wire ALL the injectors to channel 1, the engine will still get the correct fueling.

mck1117, 2023-03-28:
(think about why "two wire batch" and "one wire batch" use the same injection duration)

rusefillc, 2023-03-28:
Thank you. Hmm, tempting. Still a hack of sorts but I think a hack we can live with.

closed 2023-03-28 by rusefillc