rusefi/rusefi #5201: Pavel Mad Rat is touching FWD VQ engine

rusefillc, 2023-03-26:
https://rusefi.com/online/view.php?msq=1433

https://rusefi.com/online/view.php?log=1206
https://rusefi.com/online/view.php?log=1207

rusefillc, 2023-03-26:
https://en.wikipedia.org/wiki/Marussia_B-Series

rusefillc, 2023-03-26:
trigger: engine phase pretty mad

[image]

rusefillc, 2023-03-26:
[image]

rusefillc, 2023-03-26:
[image]

rusefillc, 2023-03-26:
See also https://github.com/rusefi/hellen121nissan-issues/issues/26

rusefillc, 2023-03-26:
See https://github.com/rusefi/rusefi/issues/5202

rusefillc, 2023-03-26:
https://rusefi.com/online/view.php?log=1208 much better with inverted CAM

[image]

closed 2023-03-27 by rusefillc

omidgheitani, 2023-03-28:
https://rusefi.com/online/view.php?msq=1433
https://rusefi.com/online/view.php?log=1206
https://rusefi.com/online/view.php?log=1207